ROBOTIS-GIT/turtlebot3_simulations #7: Realsense frame incorrect

allenh1, 2017-07-13:
Everything works pretty well, but the depth camera seems to be in the wrong frame.

Here's the view in RVIZ:

[image]

routiful, 2017-07-14:
Hi :)

I tried to change frame using tf but it does not works.
It needs additional publisher to change orientation of frame.
I attached some [reference link](http://answers.gazebosim.org/question/4266/gazebo-15-libgazebo_openni_kinectso-pointcloud-data-is-rotated-and-flipped/)

If you have any idea to change frame, please notice us.
Thanks.

allenh1, 2017-07-24:
@routiful I think I'll go ahead add a static transform publisher to the fake turtlebot node. That should resolve this.

routiful, 2017-08-02:
Hi @allenh1 

I add static transform publisher in turtlebot3_world.launch file.
You can check it in master branch.
[image]

Thanks.

allenh1, 2017-08-02:
Excellent! Nice work.

closed 2017-08-02 by routiful

cs408-ta, 2019-09-27:
I see that static transform publisher is now removed from turtlebot3_world.launch file. Can you please explain why is it not required anymore?

Thanks!

sayali-purdue, 2019-09-27:
I see that static transform publisher is now removed from turtlebot3_world.launch file. Can you please explain why is it not required anymore?

Thanks!