ROBOTIS-GIT/turtlebot3_simulations #19: No transform when simulating TB3 burger in gazebo/rviz

tropicdome, 2017-12-20:
When I start the simulation of TB3 burger following the guide on the website (http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#turtlebot3-fake-node-implementation), it says in rviz that no transform is found for the robot. Furthermore, it does not react to the teleop commands in gazebo/rviz.
[screen shot](http://en.robotis.com/BlueAD/upload/old_qna/1513773439_0.png)

Am I missing a command not mentioned in the guide or is it some other issue?

tropicdome, 2017-12-20:
I figured out what was wrong. In this commit in the burger xacro description, **publishWheelTF** has been set to false, and that breaks the simulation in gazebo/rviz
https://github.com/ROBOTIS-GIT/turtlebot3/commit/23dc9ba42ccdaa7396a2ba1ea7c1df66a19d9e47#diff-2ebbbc146f499dbcea23d40f58ee4ae8

I furthermore get these warnings when launching gazebo with the burger model

```
[ WARN] [1513776666.108005168, 0.327000000]: DiffDrive(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1513776666.109134521, 0.327000000]: DiffDrive(ns = //): <tf_prefix> = 
[ WARN] [1513776666.109938465, 0.327000000]: DiffDrive(ns = //): missing <odometrySource> default is 1
[ WARN] [1513776666.110151399, 0.327000000]: GazeboRosDiffDrive Plugin (ns = ) missing <publishTf>, defaults to 1
```

routiful, 2017-12-21:
Hello :)

TF for wheel is already worked by URDF and robot state publisher. 
So, I set 'publishWheelTF' to false. If it set true, TF for wheel is worked twice.

I check Fake node and Gazebo node. It is worked well.
Did you fix it after set 'publishWheelTF' to true??

Thanks
Darby

tropicdome, 2017-12-21:
Hi :)

Yes indeed, after setting publishWheelTF to true it is working now.

routiful, 2017-12-22:
Oh, I catch it. 

We provide rviz launch file for gazebo package.

```
  $ export TURTLEBOT3_MODEL=burger
  $ roslaunch turtlebot3_gazebo turtlebot3_simulation.launch
```

If you launch this file, you don't need a publishWheelTF. 
I guess you just used your Rviz. 

Thanks
Darby

closed 2018-01-15 by routiful