ROBOTIS-GIT/turtlebot3_simulations #20: turtlebot3_fake_node: Segmentation fault (core dumped)

yoshimalucky, 2017-12-26:
The node causes segmentation fault even if I execute it separately with rosrun.

robotis@ROBOTIS-JAPAN05:~$ roslaunch turtlebot3_fake turtlebot3_fake.launch -v
... logging to /home/robotis/.ros/log/8ca9ca98-e9fc-11e7-9e98-f48c5052e397/roslaunch-ROBOTIS-JAPAN05-12929.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

... loading XML file [/opt/ros/kinetic/etc/ros/roscore.xml]
... executing command param [rosversion roslaunch]
Added parameter [/rosversion]
... executing command param [rosversion -d]
Added parameter [/rosdistro]
Added core node of type [rosout/rosout] in namespace [/]
... loading XML file [/opt/ros/kinetic/share/turtlebot3_fake/launch/turtlebot3_fake.launch]
Added parameter [/tb3_model]
... loading XML file [/home/robotis/robotis_ws/src/ROBOTIS-GIT/turtlebot3/turtlebot3_bringup/launch/includes/description.launch.xml]
... executing command param [/opt/ros/kinetic/lib/xacro/xacro '/home/robotis/robotis_ws/src/ROBOTIS-GIT/turtlebot3/turtlebot3_description/urdf/turtlebot3_burger.urdf.xacro']
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
Added parameter [/robot_description]
... done importing include file [/home/robotis/robotis_ws/src/ROBOTIS-GIT/turtlebot3/turtlebot3_bringup/launch/includes/description.launch.xml]
Added node of type [turtlebot3_fake/turtlebot3_fake_node] in namespace [/]
Added parameter [/robot_state_publisher/publish_frequency]
Added node of type [robot_state_publisher/robot_state_publisher] in namespace [/]
Added node of type [rviz/rviz] in namespace [/]
started roslaunch server http://127.0.0.1:39878/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /tb3_model: burger

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    turtlebot3_fake_node (turtlebot3_fake/turtlebot3_fake_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[turtlebot3_fake_node-1]: started with pid [12950]
process[robot_state_publisher-2]: started with pid [12951]
process[rviz-3]: started with pid [12970]
[turtlebot3_fake_node-1] process has died [pid 12950, exit code -11, cmd /opt/ros/kinetic/lib/turtlebot3_fake/turtlebot3_fake_node __name:=turtlebot3_fake_node __log:=/home/robotis/.ros/log/8ca9ca98-e9fc-11e7-9e98-f48c5052e397/turtlebot3_fake_node-1.log].
log file: /home/robotis/.ros/log/8ca9ca98-e9fc-11e7-9e98-f48c5052e397/turtlebot3_fake_node-1*.log

routiful, 2018-01-15:
Did you solve this issue??
I have downloaded code in master branch to test. It is working well.

yoshimalucky, 2018-01-15:
My problem is caused by the binary installed package.

robotpilot, 2018-02-12:
TurtleBot3's software is still under development. 
We recommend that you download and build the source code.

robotpilot, 2018-04-08:
I recently update binary package by ROBOTIS. 
https://discourse.ros.org/t/ros-industrial-ci-for-package-maintainers-and-users/4390
Please let me know if you have problems again. I will close this issue.

closed 2018-04-08 by robotpilot