ROBOTIS-GIT/turtlebot3_simulations #21: Why is the simulation separeted into 3 launch files?

yoshimalucky, 2017-12-26:
I'm wondering about that there are 3 launch files:

turtlebot3_empty_world.launch
turtlebot3_simulation.launch
turtlebot3_fake.launch

Why do you separate these into 3 files? And, why do you separate into 2 packages gazebo and fake?

routiful, 2018-01-15:
Hello @yoshimalucky 

Thanks for your inquiry. I'm looking forward to give right answer to you.

We have supported simulation program in 2D and 3D. 
turtlebot3_fake.launch file open Rviz for 2D simulation. It shows appearance of Turtlebot3 and publish right topic for user.
turtlebot3_empty_world.launch file open Gazebo for 3D simulation. Gazebo supports a lot of sensors as plugin for simulation and ROS control. It helps to make SLAM and Navigation.
turtlebot3_empty_world.launch file is simple algorithm for object avoidance. 

Thanks
Darby

routiful, 2018-04-16:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-04-16 by routiful