ROBOTIS-GIT/turtlebot3_simulations #39: Cannot find model

arixrobotics, 2018-03-15:
Hi, 
I've just downloaded the new package which include Turtlebot3 House. 
However, when I run 
`$ roslaunch turtlebot3_gazebo_ros turtlebot3_world.launch`

Gazebo will start to open, but after a while I get 
`Error [parser.cc:581] Unable to find uri[model://turtlebot3_world]`

Similarly, if I try to launch the new house world;
`Error [parser.cc:581] Unable to find uri[model://turtlebot3_house]`

Launching the empty world works fine. 
So I think its missing the model files.

routiful, 2018-03-15:
Would you check the version??
turtlebot3_simulations are updated yesterday

Thanks
Darby

arixrobotics, 2018-03-15:
Looking at the CHANGELOG file, I got 0.2.4 (2018-03-14). 
This should be the latest correct?

I think there should be another folder with all the models in turtlebot3_gazebo_ros folder right? That particular folder should have the meshes, config and sdf files etc. It now missing.

routiful, 2018-03-15:
I'm sorry. My mistake. 

1. you need to clone `turtlebot3_gazebo_plugin`
https://github.com/ROBOTIS-GIT/turtlebot3_gazebo_plugin

2. add line in terminal

$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:${turtlebot3_gazebo_plugin path}/models

I will add this on emanual

Thanks
Darby

arixrobotics, 2018-03-15:
Yes that worked perfectly! Thanks!

Tejas-Shastha, 2018-05-23:
The export command mentioned above by @routiful did not work for me on Ubuntu 16.04. So after cloning the same repo, I instead used the following, which did end up working : (absolute path specific to my local of course)
`export GAZEBO_MODEL_PATH=~/catkin_ws/src/turtlebot3_gazebo_plugin/models:${GAZEBO_MODEL_PATH}`

routiful, 2018-05-25:
Hi @Tejas-Shastha 

GAZEBO_MODEL_PATH is already included in [package.xml](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/master/turtlebot3_gazebo/package.xml#L24).
So you don't need to export command.

Moreover, if you want to launch `turtlebot3_gazebo_plugin`, please refer below links

[Repo] : https://github.com/ROBOTIS-GIT/turtlebot3_gazebo_plugin

[WIKI] : http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#standalone-gazebo-plugin

closed 2018-05-31 by robotpilot