ROBOTIS-GIT/turtlebot3_simulations #43: [gazebo-1] process has died

markdole, 2018-04-20:
**roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch...** 
logging to /home/mark/.ros/log/ed3f78ba-447b-11e8-9ffd-d4bed9d3a2e0/roslaunch-mark-7464.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://218.197.194.58:40127/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://218.197.194.58:11311

process[gazebo-1]: started with pid [7486]
process[gazebo_gui-2]: started with pid [7491]
process[spawn_urdf-3]: started with pid [7496]
**[gazebo-1] process has died [pid 7486, exit code 255, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/mark/catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/empty.world __name:=gazebo __log:=/home/mark/.ros/log/ed3f78ba-447b-11e8-9ffd-d4bed9d3a2e0/gazebo-1.log].
log file: /home/mark/.ros/log/ed3f78ba-447b-11e8-9ffd-d4bed9d3a2e0/gazebo-1*.log**
[gazebo_gui-2] process has finished cleanly
log file: /home/mark/.ros/log/ed3f78ba-447b-11e8-9ffd-d4bed9d3a2e0/gazebo_gui-2*.log
^C[spawn_urdf-3] killing on exit
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/gazebo_ros/spawn_model", line 313, in <module>
    sm.callSpawnService()
  File "/opt/ros/kinetic/lib/gazebo_ros/spawn_model", line 271, in callSpawnService
    initial_pose, self.reference_frame, self.gazebo_namespace)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/gazebo_ros/gazebo_interface.py", line 28, in spawn_urdf_model_client
    rospy.wait_for_service(gazebo_namespace+'/spawn_urdf_model')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 159, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown
shutting down processing monitor...
... shutting down processing monitor complete
done

How can I solve this problem？

routiful, 2018-04-24:
Hi :)

Thank you for your inquiry. 
Would you tell me your development environments such as OS, PC, graphic driver?
It will be more help me to comment your issue.

markdole, 2018-04-24:
OS: Ubuntu16.04LTS  
ros: kinetic 
graphic driver :NVIDIA GT530

routiful, 2018-04-25:
Thanks a lot your kind comment. I attached additional questions below.

1. Could you open gazebo standalone? 
`$ gazebo`

2. Did you install gazebo-ros package?
`$ sudo apt-get install ros-kinetic-gazebo-ros`

3. Have you downloaded turtlebot3 meta package? and catkin_make it? It is necessary to execute turtlebot3_simulation.
`$ git clone https://github.com/ROBOTIS-GIT/turtlebot3`

markdole, 2018-04-25:
yeah I can run gazebo and insert turtlemodel by myself , and I have run teleop in rviz and control the bot moving.

markdole, 2018-04-25:
I have restart my computer ,and it just turns out right .
Don't know why , is it possible a RAM problem?

routiful, 2018-04-26:
Because i don't have all information about your PC, I can't figure it out in your case.
But,in my experiences, after update or upgrade Ubuntu, it needs to reboot PC.

If you have any other questions, please feel comfortable to open issue.

Thanks

markdole, 2018-04-26:
Thank you for your help :grinning:

TNAV1, 2019-10-12:
Same issue roslaunch turtlebot3_gazebo turtlebot3_house.launch

... logging to /home/todd/.ros/log/5da6bf72-ed0a-11e9-a84d-e0d55ea7841f/roslaunch-avdev-17564.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://avdev:33575/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [17582]
process[gazebo_gui-2]: started with pid [17587]
process[spawn_urdf-3]: started with pid [17592]
[ INFO] [1570898050.832474907]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1570898050.833518520]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[gazebo-1] process has died [pid 17582, exit code 255, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/todd/catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_house.world __name:=gazebo __log:=/home/todd/.ros/log/5da6bf72-ed0a-11e9-a84d-e0d55ea7841f/gazebo-1.log].
log file: /home/todd/.ros/log/5da6bf72-ed0a-11e9-a84d-e0d55ea7841f/gazebo-1*.log
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'

Running
OS: Ubuntu18
ros: Melodic

Can not use $gazebo to launch.

but 
roslaunch turtlebot3_gazebo turtlebot3_house.launch
launches gazebo but no turtle bot

rjshim, 2019-10-15:
@TNAV1 

Would you test below ?

$ export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:${turtlebot3_gazebo_plugin path}/build
$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:${turtlebot3_gazebo_model path}/models

${turtlebot3_gazebo_plugin_path} should be the actual path to the plugin and model folders, such as $HOME/catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo

Test above and let me know if it still doesn't work,
Ryan

TNAV1, 2019-10-17:
@rjshim 

Thanks this worked and RVIZ now shows Turtlebot!

rjshim, 2019-10-17:
@TNAV1 

Write at the bottom of ~/.bashrc file, then you won't need to type them every time.

I will close this issue.
Ryan

closed 2019-10-17 by rjshim

TNAV1, 2019-10-17:
Any suggestions for how to get 
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

to work so that I can control Waffle with the keyboard?  Does it need similar path definitions?

rjshim, 2019-10-17:
@TNAV1 

You won't need to do extra work to use it. Just follow our eManual about teleop and let me know if it doesn't work.

TNAV1, 2019-10-18:
Following the directions at 8.2.1 
http://emanual.robotis.com/docs/en/platform/turtlebot3/teleoperation/#ps3-joystick

Works!: The problem was that I was in RViz not the terminal window when trying to control waffle.

rjshim, 2019-10-18:
@TNAV1 

Great. Let us know if you have any other questions. Ryan

andy0811, 2019-10-24:
i have a problem.can you help me?[ERROR] [1571900114.497066179]: An exception was thrown: open: No such file or directory
[turtlebot3_lds-2] process has died [pid 4269, exit code 255, cmd /home/pi/catkin_ws/devel/lib/hls_lfcd_lds_driver/hlds_laser_publisher __name:=turtlebot3_lds __log:=/home/pi/.ros/log/a1568d96-f62a-11e9-a737-b827eb2c7ad7/turtlebot3_lds-2.log].
log file: /home/pi/.ros/log/a1568d96-f62a-11e9-a737-b827eb2c7ad7/turtlebot3_lds-2*.log

tasashi0805, 2021-03-21:
funny002@funny002-VirtualBox:~/catkin_ws$ roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
... logging to /home/funny002/.ros/log/fe7bdfd8-8a2d-11eb-874f-080027d2acdd/roslaunch-funny002-VirtualBox-9559.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:35729/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [9579]
process[gazebo_gui-2]: started with pid [9582]
process[spawn_urdf-3]: started with pid [9588]
[ INFO] [1616322904.723671498]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1616322904.725531323]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1616322904.750762682]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1616322904.755656333]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
[ INFO] [1616322905.503774204]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1616322905.514152610]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1616322905.653661615]: Physics dynamic reconfigure ready.
[ INFO] [1616322906.044021909, 0.026000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1616322906.054625434, 0.026000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1616322906.398555457, 0.026000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1616322906.399030887, 0.026000000]: Starting Laser Plugin (ns = /)
[ INFO] [1616322906.406030504, 0.026000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1616322906.425332911, 0.026000000]: Physics dynamic reconfigure ready.
[ INFO] [1616322906.505513897, 0.026000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1616322906.505688680, 0.026000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1616322906.507377393, 0.026000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1616322906.510159360, 0.026000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1616322906.511209749, 0.026000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1616322906.516450936, 0.026000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1616322906.519370599, 0.026000000]: DiffDrive(ns = //): Advertise odom on odom 
Segmentation fault (core dumped)
[gazebo-1] process has died [pid 9579, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/funny002/catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/empty.world __name:=gazebo __log:=/home/funny002/.ros/log/fe7bdfd8-8a2d-11eb-874f-080027d2acdd/gazebo-1.log].
log file: /home/funny002/.ros/log/fe7bdfd8-8a2d-11eb-874f-080027d2acdd/gazebo-1*.log
[spawn_urdf-3] process has finished cleanly
log file: /home/funny002/.ros/log/fe7bdfd8-8a2d-11eb-874f-080027d2acdd/spawn_urdf-3*.log

I am a ROS newbie.  trying to open a gazebo with any turtlebot and empty map. However, i only success to launch gazebo only. Can someone help me? 
I am using windows 10  vbox OS-ubuntu 16.04 - 4 processor  128mb video memory Enable 3D Acceleration. Graphic Card: NVIDIA GeForce GTX 1050 Ti

ROBOTIS-Will, 2021-03-23:
@tasashi0805 
Hi, thank you for your inquiry.
Please submit a new ticket as your issue seem to be irrelevant to the original one.