ROBOTIS-GIT/turtlebot3_simulations #56: [gazebo_gui-3] process has died [pid 3399, exit code 2, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient

kangbojk, 2018-06-29:
I'm trying to simulate turtlebot in gazebo
I tried both methods mentioned in here http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#simulation

When I run `roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
` on my PC
error occurs
_process[gazebo-2]: started with pid [3382]
process[gazebo_gui-3]: started with pid [3399]
process[spawn_urdf-4]: started with pid [3407]_
_/opt/ros/kinetic/lib/gazebo_ros/gzclient: 24: .: Can't open /home/kangbo/catkin_ws/devel/share/gazebo//setup.sh_

> [gazebo_gui-3] process has died [pid 3399, exit code 2, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/kangbo/.ros/log/e6670cae-7bb4-11e8-9151-4f401d799bca/gazebo_gui-3.log].
> log file: /home/kangbo/.ros/log/e6670cae-7bb4-11e8-9151-4f401d799bca/gazebo_gui-3*.log

_/opt/ros/kinetic/lib/gazebo_ros/gzserver: 30: .: Can't open /home/kangbo/catkin_ws/devel/share/gazebo//setup.sh_

> [gazebo-2] process has died [pid 3382, exit code 2, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/kangbo/catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_house.world __name:=gazebo __log:=/home/kangbo/.ros/log/e6670cae-7bb4-11e8-9151-4f401d799bca/gazebo-2.log].
> log file: /home/kangbo/.ros/log/e6670cae-7bb4-11e8-9151-4f401d799bca/gazebo-2*.log

I run the same command on my laptop, gazebo works fine with no error.

Since I got no clue with the bug, I tried the second method.
`gazebo worlds/turtlebot3_${TB3_MODEL}.world`
It worked on my PC.

Some parameters of my PC
OS: Ubuntu 16.04.4 LTS
ros: kinetic
VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RS780L [Radeon 3000]

Any suggestion is helpful, thanks.

kijongGil, 2018-07-02:
Hi :) @kangbojk
What is your gazebo version?
If your gazebo version is lower than 7.0, install gazebo version 7.0 over.

`$ sudo apt-get install gazebo7`

Thanks,
Gilbert

kangbojk, 2018-07-02:
My gazebo version is 7.0.0

kijongGil, 2018-07-02:
I think the link below will help you.

https://answers.ros.org/question/215796/problem-for-install-gazebo_ros_package/

robotpilot, 2018-07-08:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-07-08 by robotpilot

kangbojk, 2018-07-09:
Sorry for late response.
I changed the `setup_path=/usr/share/gazebo` as mentioned in the answer.
Setup can be found, but I have to launch gzclient in another terminal to get simulation output.
The problem is solved, thanks for supporting.

robotpilot, 2018-07-09:
It is good to hear back from you. I hope everything is going great.
Please, let me know if you have any questions on this issues.