ROBOTIS-GIT/turtlebot3_simulations #60: robot rotates at present localtion

Kanelankai, 2018-07-24:
Hello!
I met a problem when run "roslaunch turtlebot3_gazebo turtlebot3_simulation.launch" in gazebo.
The robot doesn't move forward and just rotate at the present location.
Does anyone have some idea about this problem?
Thank you!

Kanelankai, 2018-07-24:
I checked the Turtlebot3Drive::controlLoop() function and found that in case 3 of the "switch" function.
    case TB3_RIGHT_TURN:
      if (priv_right_joint_encoder_ == 0.0)
      {
        turtlebot3_state_num = GET_TB3_DIRECTION;
      }
      else
      {
        if (fabs(priv_right_joint_encoder_ - right_joint_encoder_) < 0.1)
        {
          turtlebot3_state_num = GET_TB3_DIRECTION;
        }
        else
        {
          updatecommandVelocity(0.0, -1 * ANGULAR_VELOCITY);
        }
      }
break;

The value of "fabs(priv_right_joint_encoder_ - right_joint_encoder_)" is always larger than 0.1. So case GET_TB3_DIRECTION will never be entered and the robot just rotates. I don't know the meaning of parameter "priv_right_joint_encoder" and " right_joint_encoder_" . Maybe there are some problems in the calculation of these parameters ?
Thank you.

routiful, 2018-07-25:
Hi :)

`priv_right_joint_encoder_` means required number of rotation of wheel and `right_joint_encoder_` means present joint states(radian)

Kanelankai, 2018-07-25:
@routiful Hello!
I downloaded the updated package just now.
I will test the new one and give you the feedback.
Thank you very much!

Kanelankai, 2018-07-25:
Hello! @routiful 
I have tested with the updated package by running "roslaunch turtlebot3_gazebo turtlebot3_simulation.launch".
Unfortunately, the new code didn't solve my problem.
So I tried with the former package and revised some lines of the Turtlebot3Drive::controlLoop() function and it worked.
Although I have no confidence about whether my revised code is correct or not, I paste the revised part as follows:

/*******************************************************************************
* Control Loop function
*******************************************************************************/
bool Turtlebot3Drive::controlLoop()
{
  static uint8_t turtlebot3_state_num = 0;
  double wheel_radius = 0.033;
  double turtlebot3_rotation = 0.0;

  turtlebot3_rotation = (rotate_angle_ * turning_radius_ / wheel_radius);

  switch(turtlebot3_state_num)
  {
    case GET_TB3_DIRECTION:           //this part also need some revision, abandoning "else if" 
      if (direction_vector_[CENTER] > front_distance_limit_)
      {
        turtlebot3_state_num = TB3_DRIVE_FORWARD;
      }

     // if (direction_vector_[CENTER] < front_distance_limit_ || direction_vector_[LEFT] < side_distance_limit_)
if (direction_vector_[LEFT] < side_distance_limit_)
      {
        priv_right_joint_encoder_ = right_joint_encoder_ - turtlebot3_rotation;
        turtlebot3_state_num = TB3_RIGHT_TURN;
      }
      //else if (direction_vector_[RIGHT] < side_distance_limit_)
if (direction_vector_[RIGHT] < side_distance_limit_)
      {
        priv_right_joint_encoder_ = right_joint_encoder_ + turtlebot3_rotation;
        turtlebot3_state_num = TB3_LEFT_TURN;
      }
      break;

    case TB3_DRIVE_FORWARD:
      updatecommandVelocity(LINEAR_VELOCITY, 0.0);
      turtlebot3_state_num = GET_TB3_DIRECTION;
      break;

    case TB3_RIGHT_TURN:
      if (priv_right_joint_encoder_ == 0.0)
      {
        turtlebot3_state_num = GET_TB3_DIRECTION;
      }
      else
      {
        if (fabs(priv_right_joint_encoder_ - right_joint_encoder_) < 0.1)**//since this line doesn't work ,
I copy and paste the contents to the "else" part.**
        {
          turtlebot3_state_num = GET_TB3_DIRECTION;
        }
        else
        {
          updatecommandVelocity(0.0, -1 * ANGULAR_VELOCITY);
///paste here
turtlebot3_state_num = GET_TB3_DIRECTION;
        }
      }
      break;

    case TB3_LEFT_TURN:
      if (priv_right_joint_encoder_ == 0.0)
      {
        turtlebot3_state_num = GET_TB3_DIRECTION;
      }
      else
      {
        if (fabs(priv_right_joint_encoder_ - right_joint_encoder_) < 0.1) **//since this line doesn't work ,
I copy and paste the contents to the "else" part.**
        {
          turtlebot3_state_num = GET_TB3_DIRECTION;
        }
        else
        {
          updatecommandVelocity(0.0, ANGULAR_VELOCITY);
///paste here
turtlebot3_state_num = GET_TB3_DIRECTION;
        }
      }
      break;

    default:
      turtlebot3_state_num = GET_TB3_DIRECTION;
      break;
  }

  return true;
}

routiful, 2018-08-02:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-02 by routiful