ROBOTIS-GIT/turtlebot3_simulations #61: Turtlebot3 wheel slip when moving in a curve

esleepybear, 2018-08-02:
Hi, Im using ROS melodic/gazebo to simulate TB3 moving in an arc. Im publishing constant lin/ang velocity:
cmd_vel:
linear.x=0.15
angular.z=0.28
Robot moves but not smoothly, it turns abruptly while starting and stopping. 
Here's a video. Im guessing it has something to do with the tb3 urdf since turtlesim works well with this trajectory. Here's a video of the behavior
[](https://youtu.be/WrE3BPGC6d8)

esleepybear, 2018-08-03:
Thanks a lot for the reply. Could you please telle what version of Ubuntu/Ros/gazebo you used? I'm using Ubuntu 18 with melodic. I'll try modifying the gazebo environment and will update here.

routiful, 2018-08-05:
Good, If you have any further question, please feel free to reopen issue or new issue.

closed 2018-08-05 by routiful