ROBOTIS-GIT/turtlebot3_simulations #63: Failed to install turtlebot3_gazebo

ryfr, 2018-10-07:
Hi, I am a newbie to ROS. I tried to install the turtlebot3_simulations with the following steps:

On Ubuntu 18.04.1 LTS

- installing full ros-melodic 
- setting up a catkin workspace
- git clone turtlebot3, turtlebot3_msgs, turtlebot3_simulations
- running catkin_make and sourcing the setup.bash

but I can only roslaunch 
turtlebot3_bringup 
turtlebot3_example
turtlebot3_navigation
turtlebot3_teleop
turtlebot3_description
turtlebot3_msgs
turtlebot3_slam 

catkin_make throws following warnings:

CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindBoost.cmake:911 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "chrono" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  /usr/share/OGRE/cmake/modules/FindOGRE.cmake:318 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:175 (find_package)
  turtlebot3_simulations/turtlebot3_gazebo/CMakeLists.txt:23 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

Is there a quick workaround for this?
Thanks for the help!

daniel9612, 2018-10-10:
Encountered same problem

kijongGil, 2018-10-16:
Hi @ryfr, @daniel9612 :)
I'm sorry for late reply.
I'm really sorry, but it does not support the melodic version yet. I will let you know when I release ros-melodic.

Thanks,
Gilbert

ryfr, 2018-10-18:
Ok, thanks. 
Which is the last supported version? 
Ros-lunar?

closed 2018-10-29 by kijongGil