ROBOTIS-GIT/turtlebot3_simulations #64: Port turtlebot3_simulations to ROS2

mkhansen-intel, 2018-10-23:
What's the status of porting this repo to ros2? I have pulled and tried to build the 'ros2' branch using colcon build but it looks like it hasn't been ported yet. Is that planned soon?

robotpilot, 2018-10-23:
Hi @mkhansen-intel,
This package has not yet been ported to ROS2. Now, we are reviewing the ROS2 Gazebo tutorial from OR. I'll let you know soon after the work is done.

mkhansen-intel, 2018-10-24:
@robotpilot - Thank you, we're looking forward to using it

mkhansen-intel, 2018-10-29:
@routiful - I see you added some commits with a tutorial for running this with RO2 Gazebo, is it ready for us to try?

mkhansen-intel, 2018-10-29:
FYI - We were able to get the robot to load in gazebo and respond to teleop commands. We did have to fix a few minor issues where there were missing semicolons at the end of RCLCPP_INFO statements but otherwise it worked.

routiful, 2018-10-30:
@mkhansen-intel  

I first commited last week. But current source has problems that TF is not properly working. 
So you can't see the model and scan data simultaneously in Rviz.
I assume that is caused by Rviz problem. 

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/73a12b7c9bbfebd9a0030bc4f2f394f64228c05e/gazebo_plugins/worlds/gazebo_ros_ray_sensor_demo.world#L22

https://github.com/ros2/rviz/issues/332

But `/tb3/odom`, `/tb3/cmd_vel', and '/tb3/scan' is working.

Another problem is slip in Gazebo. This problem is used to in last Gazebo version.
I added some ode setup but it is not solved yet

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/73a12b7c9bbfebd9a0030bc4f2f394f64228c05e/gazebo_plugins/worlds/gazebo_ros_ray_sensor_demo.world#L22

I am going to try more and my goal is that launch cartographer in gazebo.
If it works better, I will notice in this thread.

Thanks

mkhansen-intel, 2018-11-05:
@routiful - do you have a status update on this? We're seeing strange behavior when running the TB3 in Gazebo.

One problem we're seeing is that the Z values seem to be changing due to Gravity. I think this might be a Gazebo problem but I'm not sure?

```
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$     ros2 run tf2_ros tf2_echo odom base_link
At time 43.440000000
- Translation: [-2.000, 0.000, -9246.152]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 44.410000000
- Translation: [-2.000, 0.000, -9663.698]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 45.410000000
- Translation: [-2.000, 0.000, -10103.811]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 46.390000000
- Translation: [-2.000, 0.000, -10544.630]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
```
The Z values are rapidly changing in the negative direction, which seems like 9.8m/s squared.

We also see base_link to right_wheel and left_wheel changing in the same way, when these should be static shouldn't they?
```
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$ ros2 run tf2_ros tf2_echo base_link right_wheel
At time 72.680000000
- Translation: [-0.073, 0.039, 25883.228]
- Rotation: in Quaternion [0.007, 0.011, -0.519, 0.855]
At time 73.620000000
- Translation: [-0.074, 0.040, 26557.081]
- Rotation: in Quaternion [0.007, 0.011, -0.525, 0.851]
At time 74.620000000
- Translation: [-0.074, 0.041, 27283.452]
- Rotation: in Quaternion [0.007, 0.011, -0.531, 0.847]
At time 75.600000000
- Translation: [-0.075, 0.042, 28004.804]
- Rotation: in Quaternion [0.007, 0.011, -0.537, 0.843]
```

mkhansen-intel, 2018-11-05:
I should also mention that we see the turtlebot slowly rotating, I'm not sure if that is due to the inertia settings in the wheels?

mkhansen-intel, 2018-11-06:
I discovered that if I remove this line from the demo world file, the accelerating Z behavior stops:
<link name="base_footprint"/>

mkhansen-intel, 2018-11-06:
Also, found the same behavior with the gazebo demo, and filed an issue here:
https://github.com/ros-simulation/gazebo_ros_pkgs/issues/836

routiful, 2018-11-13:
Hi, sorry for late reply.
I accetped PR #66 and ready to test it.
If any futher progress, I will let you know.

mkhansen-intel, 2018-11-14:
@routiful - that PR helped with the slippage issue and got rid of the base_footprint, which is good. Can you add in the transforms for base_link->base_footprint->base_scan?

routiful, 2018-11-15:
Turtlebot3 has TF tree like `base_footprint->base_link->base_scan`
I added `base_footprint` again because it is not necessary but I don't want to make difference compared by ROS1 version.

Furthermore, I added imu and joint_state topic. But there is some issue.

First, TF of left_wheel_joint and right_wheel_joint has problem. The relative position is different I set in sdf. I debugged in gazebo_ros_diff_driver.cpp but I can't find problem about it.

Second, I can't see the scan data in Rviz. I don't want to say that  this problem is caused by 100% depend on rviz. Maybe I miss something.

I continue debug those and commit if I got solution.

routiful, 2018-11-19:
@mkhansen-intel 

I make a load TB3 and lidar data on Rviz2.
You can use this package to your project.

The manual is written in readme file inside of turtlebot3_gazebo package.
https://github.com/ROBOTIS-GIT/turtlebot3_simulations/tree/ros2/turtlebot3_gazebo

If you have any issues or improvements, please comments freely.

From now on, I will make an example of cartographer to use this.

[image]

mkhansen-intel, 2018-11-20:
@routiful - thank you for your work on this. I tried to build this today against the latest ROS2 master branches and the build fails. I don't know if you have tried this. I'm building ros2 from source using the directions here: https://index.ros.org/doc/ros2/Linux-Development-Setup/
I'm using the ros2 repos file from the 'master' branches (because I need them for Navigation):
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos

This is the error I get. 
```
turtlebot@turtlebot:~/turtlebot3_ws$ colcon build --symlink-install
Starting >>> turtlebot3_description
Starting >>> turtlebot3_msgs
Starting >>> pcl_conversions
Starting >>> turtlebot3_cartographer                                                   
Finished <<< turtlebot3_cartographer [0.84s]                                                                   
Starting >>> turtlebot3_simulations
Finished <<< turtlebot3_description [1.05s]                                                             
Starting >>> turtlebot3_teleop
Finished <<< pcl_conversions [2.70s]                                                                  
Starting >>> turtlebot3_bringup
Finished <<< turtlebot3_simulations [2.21s]                                                             
Finished <<< turtlebot3_bringup [0.63s]                                                             
Finished <<< turtlebot3_teleop [4.13s]                                                            
Finished <<< turtlebot3_msgs [6.18s]                       
Starting >>> turtlebot3_node
--- stderr: turtlebot3_node                             
/home/turtlebot/turtlebot3_ws/src/turtlebot3/turtlebot3_node/src/joint_states_publisher.cpp:26:48: fatal error: turtlebot3_msgs/msg/sensor_state.hpp: No such file or directory
compilation terminated.
/home/turtlebot/turtlebot3_ws/src/turtlebot3/turtlebot3_node/src/odometry_publisher.cpp:25:48: fatal error: turtlebot3_msgs/msg/sensor_state.hpp: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/joint_states_publisher.dir/src/joint_states_publisher.cpp.o] Error 1
make[2]: *** [CMakeFiles/odometry_publisher.dir/src/odometry_publisher.cpp.o] Error 1
make[1]: *** [CMakeFiles/joint_states_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/odometry_publisher.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< turtlebot3_node	[ Exited with code 2 ]

Summary: 7 packages finished [7.50s]
  1 package failed: turtlebot3_node
  1 package had stderr output: turtlebot3_node
turtlebot@turtlebot:~/turtlebot3_ws$ 
```
I wasn't getting this last week, so I suspect something in ROS2 upstream repos changed, but wanted to ask you if you can reproduce.

routiful, 2018-11-20:
Of course, I will re-clone ros2.repos by master branch and test it.
It might be take one or two days because of other works. 
For now, you can make AMENT_IGNORE and COLCON_IGNORE file in ros2 package to pass it when colcon build.

I will let you know when update the turtlebot3 code. 
Thanks

mkhansen-intel, 2018-11-20:
Interestingly, I was able to get it to build on my 18.04 system, but not on the turtlebot3 itself, which is 16.04. Thanks for taking a look. We're really trying hard to use Turtlebot3 for Navigation2.

mkhansen-intel, 2018-11-20:
Also, I still don't see the base_link->base_scan transform showing up in RVIZ or when I view the /tf topic. Should that be there?

routiful, 2018-11-21:
@mkhansen-intel 

Great. I have been working in 18.04 system. And master branch turtlebot3 is built.
And base_link->base_scan tf is presented in rviz2. You can check it `tf2_echo` command
Because this is static tf, the time is 0.0

```
ost@demo:~$ ros2 run tf2_ros tf2_echo base_link base_scan
At time 0.0
- Translation: [-0.032, 0.000, 0.172]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 0.0
- Translation: [-0.032, 0.000, 0.172]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 0.0
- Translation: [-0.032, 0.000, 0.172]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 0.0
- Translation: [-0.032, 0.000, 0.172]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
```

By the way, I open a issue for problem of tf stamp (https://github.com/ros-simulation/gazebo_ros_pkgs/issues/840) and it is solved now. But I can't find to run occupancy_grid_node in cartographer_ros.

mkhansen-intel, 2018-11-21:
@routiful - I don't see that transform:
mkhansen@mkhansen-desk:~/ros2_dev/turtlebot3_ws$ ros2 run tf2_ros tf2_echo base_link base_scan
Failure at 1.54283e+09
Exception thrown:"base_link" passed to lookupTransform argument target_frame does not exist. 
The current list of frames is:
Frame base_footprint exists with parent odom.

routiful, 2018-11-22:
Oh, you should launch remote.launch

```
$ ros2 launch turtlebot3_bringup turtlebot3_robot.launch.py
```

mkhansen-intel, 2018-11-22:
@routiful - so I need the robot.launch to run while also running gazebo?

routiful, 2018-11-23:
Yes, libgazebo_ros_diff_drive library in gazebo_pks is to make only odom tf.
If you need all tf tree, you are supposed to launch robot_state_publisher package.

mkhansen-intel, 2018-11-25:
@routiful - now I am trying to run on the actual turtlebot3. We are using a 'waffle' model with the Joule board. I'm looking at the directions here: http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#ros2

Can these directions work for the 'waffle' even though the opencr update is for the 'burger'?

If not, can we get an opencr update for waffle too?

mkhansen-intel, 2018-11-26:
Update - I just tried running with the ros1_bridge and that won't work. The bridge doesn't know how to bridge the tf/tfMessage ROS1 msg to the tf2_msgs/TFMessage ROS2 msg. I would really appreciate an openCR update for the waffle.

mkhansen-intel, 2018-11-29:
Hi @routiful, sorry to be a pest, can you **please** tell me when we could have an opencr update for the waffle? :)

routiful, 2018-12-01:
Sorry for late reply.
We have being prepared new firmware compatible with Micro-XRCE-DDS(https://github.com/eProsima/Micro-XRCE-DDS) (github : https://github.com/ROBOTIS-GIT/ros2arduino) to merge ROS2 crystal. 
I think we need more time to announce new firmware for TB3.

So, I let you know last firmware based on micrortps to compatible with Waffle.
You need to clone OpenCR library in feature-ros2-micrortps. And add hardware. Below link might be help you.

https://community.robotsource.org/t/howto-adding-multiple-opencr-boards-to-the-arduino-ide/1497

And open turtlebot3_ros2.ino and modify parameter for waffle

https://github.com/ROBOTIS-GIT/OpenCR/blob/488827a424642fc3c1f27c0b4caae102d57af3e0/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_burger.h#L25

If you have any problem to upload firmware, please comment to me.

mkhansen-intel, 2018-12-05:
@routiful - I have cloned the OpenCR repo, installed Arduino IDE, added the OpenCR library, and I'm trying to build the turtlebot3_burger now to make sure that can build before changing it to turtlebot3_waffle. I'm getting build errors that it can't find the fastrtps.hpp file. 
```
Arduino: 1.8.7 (Linux),
Board: "OpenCR Board, OpenCR Bootloader"

/home/mkhansen/arduino-1.8.7/arduino-builder -dump-prefs -logger=machine -hardware /home/mkhansen/arduino-1.8.7/hardware -hardware /home/mkhansen/.arduino15/packages -tools /home/mkhansen/arduino-1.8.7/tools-builder -tools /home/mkhansen/arduino-1.8.7/hardware/tools/avr -tools /home/mkhansen/.arduino15/packages -built-in-libraries /home/mkhansen/arduino-1.8.7/libraries -libraries /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10807 -build-path /tmp/arduino_build_596454 -warnings=none -build-cache /tmp/arduino_cache_754426 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_gcc.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_gcc-5.4.0-2016q2.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_tools.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_tools-1.0.0.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -verbose /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino
/home/mkhansen/arduino-1.8.7/arduino-builder -compile -logger=machine -hardware /home/mkhansen/arduino-1.8.7/hardware -hardware /home/mkhansen/.arduino15/packages -tools /home/mkhansen/arduino-1.8.7/tools-builder -tools /home/mkhansen/arduino-1.8.7/hardware/tools/avr -tools /home/mkhansen/.arduino15/packages -built-in-libraries /home/mkhansen/arduino-1.8.7/libraries -libraries /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10807 -build-path /tmp/arduino_build_596454 -warnings=none -build-cache /tmp/arduino_cache_754426 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_gcc.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_gcc-5.4.0-2016q2.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_tools.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_tools-1.0.0.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -verbose /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino
Using board 'OpenCR' from platform in folder: /home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2
Using core 'arduino' from platform in folder: /home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2
Detecting libraries used...
/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10807 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/bsp/opencr -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/bsp/opencr/include -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw/driver -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw/usb_cdc -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/cores/arduino -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR /tmp/arduino_build_596454/sketch/turtlebot3_core.ino.cpp -o /dev/null
/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10807 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/bsp/opencr -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/bsp/opencr/include -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw/driver -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw/usb_cdc -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/Open [... cut ...]
Using library turtlebot3_ros_lib in folder: /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros_lib (legacy)
Using library ROS2 in folder: /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries/ROS2 (legacy)
exit status 1
Error compiling for board OpenCR Board.
```
I know the file is here:
```
mkhansen@mkhansen-desk:~/OpenCR/arduino/opencr_arduino/opencr$ find . -name micrortps.hpp
./variants/OpenCR/lib/micrortps/micrortps.hpp
```
I'm not very experienced with Arduino, so don't know how to point it to the right file. Please help.

mkhansen-intel, 2018-12-05:
Update: I was able to get past the issue above by creating a symbolic link:
```
mkhansen@mkhansen-desk:~/OpenCR/arduino/opencr_arduino/opencr$ ln -s $PWD/variants/OpenCR/lib/micrortps libraries/micrortps 
mkhansen@mkhansen-desk:~/OpenCR/arduino/opencr_arduino/opencr$ ll libraries/micrortps 
lrwxrwxrwx 1 mkhansen mkhansen 81 Dec  5 12:48 libraries/micrortps -> /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/variants/OpenCR/lib/micrortps/
```
Now it fails to find the microcdr.h header file:
```
Arduino: 1.8.7 (Linux), Board: "OpenCR Board, OpenCR Bootloader"

/home/mkhansen/arduino-1.8.7/arduino-builder -dump-prefs -logger=machine -hardware /home/mkhansen/arduino-1.8.7/hardware -hardware /home/mkhansen/.arduino15/packages -tools /home/mkhansen/arduino-1.8.7/tools-builder -tools /home/mkhansen/arduino-1.8.7/hardware/tools/avr -tools /home/mkhansen/.arduino15/packages -built-in-libraries /home/mkhansen/arduino-1.8.7/libraries -libraries /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10807 -build-path /tmp/arduino_build_810850 -warnings=none -build-cache /tmp/arduino_cache_251445 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_gcc.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_gcc-5.4.0-2016q2.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_tools.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_tools-1.0.0.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -verbose /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino
/home/mkhansen/arduino-1.8.7/arduino-builder -compile -logger=machine -hardware /home/mkhansen/arduino-1.8.7/hardware -hardware /home/mkhansen/.arduino15/packages -tools /home/mkhansen/arduino-1.8.7/tools-builder -tools /home/mkhansen/arduino-1.8.7/hardware/tools/avr -tools /home/mkhansen/.arduino15/packages -built-in-libraries /home/mkhansen/arduino-1.8.7/libraries -libraries /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries -fqbn=OpenCR:OpenCR:OpenCR:upload_method=UploadMethod -ide-version=10807 -build-path /tmp/arduino_build_810850 -warnings=none -build-cache /tmp/arduino_cache_251445 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.opencr_gcc.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_gcc-5.4.0-2016q2.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2 -prefs=runtime.tools.opencr_tools.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -prefs=runtime.tools.opencr_tools-1.0.0.path=/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_tools/1.0.0 -verbose /home/mkhansen/OpenCR/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino
Using board 'OpenCR' from platform in folder: /home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2
Using core 'arduino' from platform in folder: /home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2
Detecting libraries used...
/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10807 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/bsp/opencr -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/bsp/opencr/include -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw/driver -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw/usb_cdc -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/lib/STM32F7xx_HAL_Driver/Inc/ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/cores/arduino -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR /tmp/arduino_build_810850/sketch/turtlebot3_core.ino.cpp -o /dev/null
/home/mkhansen/.arduino15/packages/OpenCR/tools/opencr_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -mfloat-abi=softfp -mfpu=fpv5-sp-d16 -w -DDEBUG_LEVEL=DEBUG_NONE -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_OpenCR -w -x c++ -E -CC -mcpu=cortex-m7 -DF_CPU=216000000L -DARDUINO=10807 -DARDUINO_OpenCR -DARDUINO_ARCH_OPENCR -mthumb -DSTM32F746xx -D__OPENCR__ -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/bsp/opencr -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/bsp/opencr/include -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2/variants/OpenCR/hw/driver -I/home/mkhansen/.arduino15/packages/OpenCR/hardware/OpenCR/1.3.2 [... cut ...]
mkhansen@mkhansen-desk:~/OpenCR$ find . -name microcdr*
./arduino/opencr_arduino/opencr/variants/OpenCR/lib/micrortps/user_config/microcdr
mkhansen@mkhansen-desk:~/OpenCR$ ll ./arduino/opencr_arduino/opencr/variants/OpenCR/lib/micrortps/user_config/microcdr
total 12
drwxr-xr-x 2 mkhansen mkhansen 4096 Dec  4 15:08 ./
drwxr-xr-x 5 mkhansen mkhansen 4096 Dec  4 15:08 ../
-rwxr-xr-x 1 mkhansen mkhansen  951 Dec  4 15:08 config.h*
mkhansen@mkhansen-desk:~/OpenCR$ 
```
I do find the microcdr directory with a config.h file in it, but no microcdr.h file.

routiful, 2018-12-06:
You shoud git pull submodule to pull micrortps and microcdr.
I attached link for you

https://stackoverflow.com/questions/1030169/easy-way-to-pull-latest-of-all-git-submodules

routiful, 2018-12-12:
Did you solve that issue?
We are ready for release new firmware with Micro-XRCE-DDS.
I think next week you can use that.

closed 2019-02-27 by gpwhd07