ROBOTIS-GIT/turtlebot3_simulations #74: Gazebo crashes on launching of various TB3 simulation commands

pitosalas, 2019-02-07:
After a clean install of Ubuntu and the whole TB3 stack, I still get this crash. Note that this happens for any of the provided launch files.

`$ roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch`

`[gazebo_gui-2] process has died [pid 14507, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/pitosalas/.ros/log/20092ab8-2a70-11e9-b967-002500efe1ca/gazebo_gui-2.log].
`
I've seen many people report this but no answers!

routiful, 2019-02-07:
Sorry for inconvenience.
If your PC didn't get graphic driver, you can't open gazebo world.
Please check your graphic driver first.

Could you try open gazebo alone?

pitosalas, 2019-02-07:
How do I check my graphic driver?

Also gazebo alone doesn't work either.

Also see:
http://answers.gazebosim.org/question/21894/gazebo-crashes-regularly-and-somehow-this-is-related-to-another-package-of-mine/
for more info.

-- Pito

routiful, 2019-02-07:
You can use below commands for check your graphic card

```
$ lspci | grep VGA
```

If your PC was installed nvidia graphic card, you can use below as well

```
$ nvidia-settings
```

After find your graphic card, you can download proper driver on your PC.

pitosalas, 2019-02-08:
Here's what the commands says:

NVIDIA Corporation G96 [GeForce 9500 GT] (rev a1)

Now what? How are you sure its a driver? It was working fine a while ago
and the hardware hasnt changed!
-
- Pito

routiful, 2019-02-08:
The graphic driver was installed correctly, Gazebo should be openned.
How to install gazebo previously? Did you used desktop-full install command?

pitosalas, 2019-02-09:
Yes, I did do that install.

I find that even if I exit Gazebo with the menu exit command, it takes a long time to finally return to the shell.

I also find that sometimes there are multiple processes left over after it has exited. 

Is that typical or does it tell you anything?

Thanks!

Pito Salas
Brandeis Computer Science
Volen 134

routiful, 2019-02-11:
Exit bug has existed yet.
So I used to special command to kill server and client

```
$ killall -9 gazebo & killall -9 gzserver  & killall -9 gzclient
```

closed 2019-02-27 by gpwhd07