ROBOTIS-GIT/turtlebot3_simulations #75: odom linear velocity in Y axis is very noisy

crdelsey, 2019-02-21:
I'm seeing a lot of noise in the `/odom` data when running Turtlebot 3 in Gazebo. The noise in `/odom/twist/twist/linear/y` is wreaking havoc with the local planner in the nav2 stack because, at times, the apparent sideways velocity is greater than the kinematic limits of the robot.

There is a video here showing the plot of linear/x and linear/y velocities in RQT.
https://youtu.be/NCnlEXmZRUU

The most interesting bit is at 2:00 where the Turtlebot has been driving steadily for a while when all of a sudden a large but noisy Y velocity appears.

I thought this may be related to physics parameters of the Turtlebot model due to a similar issue reported in ros-simulation/gazebo_ros_pkgs#245

Details:
Running on ROS2, Gazebo 9. Ubuntu 18.04,
Running ROS2 HEAD of this repo (198218df92)

routiful, 2019-02-27:
I didn't know about that.
I will be check it.

routiful, 2019-07-01:
@crdelsey As shown other branches, it might be fixed. If it is, please close this issue.

rjshim, 2019-10-15:
As there have been no updates, I am closing this issue.

closed 2019-10-15 by rjshim