ROBOTIS-GIT/turtlebot3_simulations #80: ResourceNotFound: turtlebot3_description

hamkuchan, 2019-06-10:
when I run roslaunch turtlebot3_gazebo turtlebot3_world.launch, then occur the error:
ResourceNotFound: turtlebot3_description.
And I could find the file "turtlebot3_description" anywhere.

ROBOTIS-Kayman, 2019-06-10:
I think you miss export the turtlebot3 model.
If you use a burger, please type like the below before running the gazebo.
```
$ export TURTLEBOT3_MODEL=burger
```

Please refer to this links
 - [simulation](http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#simulate-in-various-world)
 - [export TB3 MODEL](http://emanual.robotis.com/docs/en/platform/turtlebot3/export_turtlebot3_model/)

Regards,
Kayman

hamkuchan, 2019-06-10:
> I think you miss exporting the turtlebot3 model.
> If you use a burger, please type like the below before running the gazebo.
> 
> ```
> $ export TURTLEBOT3_MODEL=burger
> ```
> 
> Please refer to this links
> 
>     * [simulation](http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#simulate-in-various-world)
> 
>     * [export TB3 MODEL](http://emanual.robotis.com/docs/en/platform/turtlebot3/export_turtlebot3_model/)
> 
> 
> Regards,
> Kayman

Hi, I forgot to download the turtlebot3 package, sorry for that.
but also thanks!

closed 2019-06-10 by ROBOTIS-Kayman