ROBOTIS-GIT/turtlebot3_simulations #81: Failed to run gazebo with turtlebot3_house.launch

tfe104270, 2019-07-01:
Hi,

I want to run turtlebot3(burger) gazebo simulation, and execute the following command according to wiki:
`roslaunch turtlebot3_gazebo turtlebot3_house.launch`

And the gazebo hang in the "preparing your world" message.
The console message is shown below:

> ll@oa:~/catkin_turtle$ roslaunch turtlebot3_gazebo turtlebot3_house.launch
> ... logging to /home/ll/.ros/log/4c26f992-9bc0-11e9-a02a-aced5ca8ad3b/roslaunch-oa-12987.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://192.168.43.77:46469/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /robot_description: <?xml version="1....
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
>  * /use_sim_time: True
> 
> NODES
>   /
>     gazebo (gazebo_ros/gzserver)
>     gazebo_gui (gazebo_ros/gzclient)
>     spawn_urdf (gazebo_ros/spawn_model)
> 
> ROS_MASTER_URI=http://192.168.43.77:11311
> 
> process[gazebo-1]: started with pid [13009]
> process[gazebo_gui-2]: started with pid [13014]
> process[spawn_urdf-3]: started with pid [13019]
> ros.gazebo_ros.api_plugin: Finished loading Gazebo ROS API Plugin.
> ros.roscpp: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
> ros.gazebo_ros.api_plugin: Finished loading Gazebo ROS API Plugin.
> ros.roscpp: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
> 

I tried gazebo v7.0 and v7.15, and the phenomenon is the same.
I can run the following command and successfully load the world model
`roslaunch turtlebot3_gazebo turtlebot3_world.launch`
`roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch`

And I also tried several models from gazebo website, these models were loaded successfully :
`https://bitbucket.org/osrf/gazebo_models/src/default/`

My environment is
    - Ubuntu 16.04.6, 
    - ros kinetic

Please give me some comment, thanks!

ROBOTIS-Kayman, 2019-07-01:
Hi,
Because gazebo downloads the environment, it takes a few minutes to run `house.launch` for the first time.
In my case, it takes 2-3 minutes.

Please run it again and wait for 2~3 minutes.

Regards, 
Kayman

tfe104270, 2019-07-01:
Hi Kayman,

It works, thank you.
One suggestion is could you please add this recommendation to turtlebot wiki so that people won't stuck in such situation.

closed 2019-07-01 by tfe104270