ROBOTIS-GIT/turtlebot3_simulations #83: ROS2: How to launch multiple TB3 robots in single world in gazebo

kishornaik10, 2019-07-31:
Hi,

I want to launch multiple TB3 robots in single world in gazebo simulator in ros2 environment.
How this can be done? please help

routiful, 2019-08-01:
@kishornaik10 
You can refer to below launch file
https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/master/turtlebot3_gazebo/launch/multi_turtlebot3.launch

kishornaik10, 2019-08-01:
@routiful 

Thank you,
This launch file can be launch in ROS (i.e. melodic) but mine case is ROS2 (i.e. dashing).
[Github repo]
https://github.com/ROBOTIS-GIT/turtlebot3/tree/ros2 
https://github.com/ROBOTIS-GIT/turtlebot3_simulations/tree/ros2 

Can you suggest modification needed in launch files in ros2? or any sample example.
Thanks 
Kishor

routiful, 2019-08-02:
@kishornaik10 
I didn't try to make launch file loading multiple TB3 in ROS2 yet.
But it seems to be not difficult to port it. 
Please feel free to comments while you have any problem when you port it :)

routiful, 2019-09-08:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-09-08 by routiful