ROBOTIS-GIT/turtlebot3_simulations #91: Add "--ros-args -r" arguments for Eloquent

mkhansen-intel, 2019-09-11:
The tb3 model files will need updating for eloquent to add the "--ros-args -r" to the arguments list.

See issue here:
ros-simulation/gazebo_ros_pkgs#991

routiful, 2019-09-16:
@mkhansen-intel 
I've tested this repo with Dashing patch 3 and every topic are shown correctly.

```
/clock
/cmd_vel
/imu
/joint_states
/odom
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
```

I think this updates doesn't need to yet. But for next version of ROS 2, I will make another branch to support this issue.

mkhansen-intel, 2019-09-16:
Correct, this is not needed for Dashing, but is required for Eloquent. I just wanted you to be aware.

mkhansen-intel, 2020-01-13:
Any update on this? Are you planning to do an Eloquent release at some point soon?

CleopatraDion, 2020-04-02:
Yes, I am also curious if this is going to be done, or what specifically needs to be done?

ROBOTIS-Will, 2020-05-11:
Sorry for the delayed update. I know we are all looking for the Foxy at this point, but please let me know if this doesn't work.

mkhansenbot, 2020-05-14:
I can't test it but thanks for merging a fix. I'll close this issue. Will there be a Foxy branch created?

closed 2020-05-14 by mkhansenbot

ROBOTIS-Will, 2020-05-14:
@mkhansenbot Yes, turtlebot3_simulations foxy branch will be created and updated along with turtlebot3 repository. Thank you!