ROBOTIS-GIT/turtlebot3_simulations #93: No /cms_vel when launching turtlebot3_world.launch

shivaang12, 2019-09-20:
In ROS1, /cmd_vel topic was used to publish the velocities for turtlebot3. But when doing the same here it doesn't work. I am wondering if its the bug or there is any other way to achieve the same thing?

Thank you in advance!
P.S. I am using Dashing.

rjshim, 2019-09-21:
@shivaang12 

Would you try below and take & show us a screenshot of your terminal window? 
$ ros2 topic list 

Regards,
Ryan

shivaang12, 2019-09-21:
@rjshim 

Thanks for the reply! I have attached the screenshot as requested.

Best,
Shivang

[image]

rjshim, 2019-09-22:
@shivaang12 

Check if you have a burger model file of the latest version (path to the model file can be checked at [link](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/ros2/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf))

You can also just redownload the entire package by doing below.
$ https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git -b ros2 

Let me know the result, 
Ryan

shivaang12, 2019-09-22:
@rjshim 

Thanks for the suggestion! I updated the burger model and it worked!

I am closing this issue.

Best,
Shivang

closed 2019-09-22 by shivaang12

rjshim, 2019-09-23:
@shivaang12 

Great! Just in case, you should type 'git clone http://~' to download a package. I wrote it wrong in the past answer.

Let us know if you have any other questions, 
Ryan