ROBOTIS-GIT/turtlebot3_simulations #94: turtlebot3_simulation.launch failed possibly related to `${GAZEBO_LIBRARIES}`

xpharry, 2019-09-30:
```
➜  turtlebot3_simulations git:(master) roslaunch turtlebot3_gazebo turtlebot3_simulation.launch
... logging to /home/pxu/.ros/log/9ebc8ea2-e3d6-11e9-95f9-d89ef348ac6b/roslaunch-sv870169lx-3667.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sv870169lx:44192/

SUMMARY
========

PARAMETERS
 * /cmd_vel_topic_name: /cmd_vel
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    turtlebot3_drive (turtlebot3_gazebo/turtlebot3_drive)

ROS_MASTER_URI=http://localhost:11311

process[turtlebot3_drive-1]: started with pid [3689]
================================================================================REQUIRED process [turtlebot3_drive-1] has died!
process has died [pid 3689, exit code -11, cmd /home/pxu/projects/turtlebot3_challenge/catkin_ws/devel/lib/turtlebot3_gazebo/turtlebot3_drive __name:=turtlebot3_drive __log:=/home/pxu/.ros/log/9ebc8ea2-e3d6-11e9-95f9-d89ef348ac6b/turtlebot3_drive-1.log].
log file: /home/pxu/.ros/log/9ebc8ea2-e3d6-11e9-95f9-d89ef348ac6b/turtlebot3_drive-1*.log
Initiating shutdown!
================================================================================
[turtlebot3_drive-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

It seems that there is a problem in the following line:

```
target_link_libraries(turtlebot3_drive ${catkin_LIBRARIES} ${GAZEBO_LIBRARIES})
```
until I modify it to

```
target_link_libraries(turtlebot3_drive ${catkin_LIBRARIES})
```

Probably it is because some library missing in my system. Can you please give me some hint?

rjshim, 2019-10-10:
@xpharry 

Can you show us your CMakeLists.txt and package.xml files in the turtlebot3_gazebo package? Also, let us know how you installed ROS Kinetic. 

Regards,
Ryan

xpharry, 2019-10-11:
https://www.dropbox.com/s/ab5uokt4jv9tr6m/CMakeLists.txt?dl=0
https://www.dropbox.com/s/su6mfw5rbj5rt0y/package.xml?dl=0

I only changed the line in CMakeLists.txt that I mentioned in the question.

I installed ROS Kinetic as normal as 

`
sudo apt-get install ros-kinetic-desktop-full
`

Gazebo version is 7.0.0 accordingly.

rjshim, 2019-10-15:
@xpharry 

Maybe some dependencies are missing and you are having that issue. You can ask this question at the ROS [forum](https://answers.ros.org/questions/) but rather than doing that, I just recommend you reinstall the OS and ROS [ref](http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#install-ros-on-remote-pc)

Regards,
Ryan

rjshim, 2019-10-21:
As there have been no updates, I am closing this issue.

closed 2019-10-21 by rjshim