ROBOTIS-GIT/turtlebot3_simulations #95: Unable to pass in parameters to start gazebo worlds without gui

Naireen, 2019-10-12:
While working with this project, to make it less computationally taxing on my computer, I was trying to launch the different world files (ex, turtlebot3_world.launch) without loading the gazebo client, and only have it bring up the gz server. Looking at the turtlebot3_world.launch file, it seems like I can't do this since the only parameters that it seems to accept is

```
 <arg name="x_pos" default="-2.0"/> 
 <arg name="y_pos" default="-0.5"/>
 <arg name="z_pos" default="0.0"/>
```

the file later has 

```
<include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="gui" value="true/>
    ...
</include>
```

Is there any way of overriding it, or would this file require editing in the first block to accept another parameter at the top level, and then pass that into the include block to not launch the gazebo client?

rjshim, 2019-10-15:
@Naireen 

[This](http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#standalone-gazebo-plugin) might be of your interest.

Regards,
Ryan

rjshim, 2019-10-21:
As there have been no updates, I am closing this issue.

closed 2019-10-21 by rjshim