ROBOTIS-GIT/turtlebot3_simulations #102: turtlebot3_drive does not publish to cmd_vel

svdeepak99, 2020-01-01:
I am using the melodic version of ROS with an Ubuntu 18.04 system.

I ran into an issue where the burger bot in gazebo, did not move when `turtlebot3_simulation.launch` file was launched. 

When I ran "rqt_graph" node I found out that the "turtlebot3_drive" node did not publish to /cmd_vel topic and also it did not subscribe from /odom and /scan topics too.

I made the following change in the in the "turtlebot3_drive.cpp" file present in the "/turtlebot3_gazebo/src" folder and the issue was solved. I added the `turtlebot3_drive.init();` line in the main() function as shown below:
```
int main(int argc, char* argv[])
{
  ros::init(argc, argv, "turtlebot3_drive");
  Turtlebot3Drive turtlebot3_drive;
  turtlebot3_drive.init();  //the line that I added

  ros::Rate loop_rate(125);

  while (ros::ok())
  {
    turtlebot3_drive.controlLoop();
    ros::spinOnce();
    loop_rate.sleep();
  }

  return 0;
}
```
Probably it happened because, some compilers do not run the init() function of a class, as soon as it's object is created.

I thought I could post this here, as if anyone faced the same issue as me could get it sorted.

Thank you.

ROBOTIS-Will, 2020-01-22:
@svdeepak99 
ROS_ASSERT() is same as the assert() and bypassed when `NDEBUG` is defined in the code.
In such cases, you can call `init()` function separately to avoid the issue.
We'll review the code and update.
Thank you.

JimEverest, 2020-04-13:
@svdeepak99 Thanks for your kind remind. I found the turtlebot doesn't move after launch turtlebot3_gazebo/turtlebot3_world.launch. So I found this opened issue and remind. I have added the line "  turtlebot3_drive.init(); "  in  turtlebot3_drive.cpp, then catkin_make whole work space. Now I can see the turtlebot3_drive node subscribing the "/cmd_vel" topic, however it is still not moving in gazebo with turtlebot3_teleop_key.launch.  am I missing something?

svdeepak99, 2020-04-13:
Hi Jim. Did you check if the turtlebot3_teleop_key node is publishing to the /cmd_vel topic & did you confirm if the values are being printed with the rostopic echo command?

JimEverest, 2020-04-14:
@svdeepak99 Thanks for your quick reply. I do confirm the /cmd_vel topic were published successfully by the  turtlebot3_teleop_key node after checking the rostopic echo command and it is worth to mention that the teleop node also works with the turtlebot3_fake/turtlebot3_fake.launch node, the turtlebot3 model moves in the rviz when I press W button.

ROBOTIS-Will, 2020-05-19:
@JimEverest Sorry for the late reply. I'm just checking in whether you are still having trouble with the Gazebo + teleop key. I didn't experience the issue on my machine after executing below commands.
```
$ roslaunch turtlebot3_gazebo turtlebot3_world.launch
```
```
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
```
[image]
[image]

jtisaacs, 2020-06-18:
@ROBOTIS-Will I am experiencing the same issue that @JimEverest mentioned above with the Gazebo + teleop key. I do experience the issue on my machine after executing below commands.
$ roslaunch turtlebot3_gazebo turtlebot3_world.launch
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

I am using ROS version 1.14.6 on Ubuntu 18.04.4 LTS, and like @JimEverest I am able to move the via teleop in rviz using the following commands.
$ roslaunch turtlebot3_fake turtlebot3_fake.launch
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

adipandas, 2020-08-28:
Did anyone resolve this issue? I am also facing the same problem as @jtisaacs. Rviz nodes are working but I am not able to drive turtlebot in gazebo. Any help will be appreciated.
Thank you.

ROBOTIS-Will, 2020-09-01:
@jtisaacs If you are unable to move TurtleBot3 on Gazebo, you may not installed all necessary Gazebo packages such as gazebo-plugins. Please install all gazebo packages with below command and try again. Also, please note that Kinetic works best with Ubuntu 16.04. I'd recommend using Melodic for your Ubuntu 18.04 or install Ubuntu 16.04 to use ROS 1 Kinetic.
```
$ sudo apt install ros-kinetic-gazebo*
```

@adipandas You can try installing all gazebo packages with below command. First, check which gazebo version you are running then install necessary pacakges.
```
$ gazebo --version
```
If you are running gazebo 9,
```
$ sudo apt install ros-melodic-gazebo-*
```
If you are running gazebo 11,
```
$ sudo apt install ros-melodic-gazebo11*
```

adipandas, 2020-09-01:
@ROBOTIS-Will Your suggestion solved the issue. I actually did not consider checking my gazebo packages.
Thank you. :smile: 

I think this issue can be closed now.

ROBOTIS-Will, 2020-09-02:
ROS_ASSERT issue has been fixed in #120. Please let us know if there's any issue with this fix.

@adipandas Glad to hear that your issue is gone now!

closed 2020-09-02 by ROBOTIS-Will

lyhlau, 2021-01-06:
I am using ROS Melodic and i tried this command turtlebot3_simulation.launch
After making the change like this thread (add this turtlebot3_drive.init();)
It still does not work.
Anyone able to help with this?

ROBOTIS-Will, 2021-01-07:
@lyhlau 
I just tested with the latest source code, but couldn't find the issue.
Could you give me the list of command you entered with the rqt_graph image?
Also, make sure that all your source code is up to date.
Thanks.

lyhlau, 2021-01-07:
I tried this 2 command
roslaunch turtlebot3_gazebo turtlebot3_world.launch
roslaunch turtlebot3_gazebo turtlebot3_simulation.launch

the world is able to launch but the 2nd command does not work, turtlebot not moving
I had tried the fix in the first thread here -> the turtlebot3_drive.init();

ROBOTIS-Will, 2021-01-07:
Please try with the latest code after removing the installed package.
```bash
$ sudo apt remove ros-melodic-turtlebot3 ros-melodic-turtlebot3-simulations
$ mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws/src
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git
$ cd ~/catkin_ws && catkin_make
```

lyhlau, 2021-01-07:
> Please try with the latest code after removing the installed package.
> 
> ```shell
> $ sudo apt remove ros-melodic-turtlebot3 ros-melodic-turtlebot3-simulations
> $ mkdir -p ~/catkin_ws/src
> $ cd ~/catkin_ws/src
> $ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
> $ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git
> $ cd ~/catkin_ws && catkin_make
> ```

I just tried your steps
then I run this command
roslaunch turtlebot3_gazebo turtlebot3_world.launch
[Err] [REST.cc:205] Error in REST request - there is this bar coming out in red
but the world is able to pops out with the turtlebot

but it still does not move by itself when i run this roslaunch turtlebot3_gazebo turtlebot3_simulation.launch

ROBOTIS-Will, 2021-01-07:
Looks like you have some issue with your installed Gazebo.
https://answers.gazebosim.org//question/25030/gazebo-error-restcc205-error-in-rest-request/#:~:text=As%20can%20be%20seen%20in,ignitionrobotics.org%20inside%20the%20~%2F.

lyhlau, 2021-01-07:
It does help resolve the [Err] line
but it still does not move

Code i get after roslaunch turtlebot3_gazebo turtlebot3_world.launch
============================================================
SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [2421]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a3158cd6-50b5-11eb-b3d6-0800278352f6
process[rosout-1]: started with pid [2432]
started core service [/rosout]
process[gazebo-2]: started with pid [2435]
process[gazebo_gui-3]: started with pid [2440]
process[spawn_urdf-4]: started with pid [2444]
[ INFO] [1610002666.335278181]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1610002666.338069566]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1610002666.406683405]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1610002666.409612771]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1610002668.232579584]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1610002669.085459506, 0.001000000]: Physics dynamic reconfigure ready.
[ INFO] [1610002669.086482145, 0.001000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1610002669.086710156, 0.001000000]: Starting Laser Plugin (ns = /)
[ INFO] [1610002669.090041770, 0.001000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1610002669.168699376, 0.001000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1610002669.169338188, 0.001000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1610002669.170509975, 0.001000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1610002669.172321674, 0.001000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1610002669.173231288, 0.001000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1610002669.176462772, 0.001000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1610002669.177500295, 0.001000000]: DiffDrive(ns = //): Advertise odom on odom 
[spawn_urdf-4] process has finished cleanly
log file: /home/lau/.ros/log/a3158cd6-50b5-11eb-b3d6-0800278352f6/spawn_urdf-4*.log
============================================================

It seems normal and succeeded
Then codes i get after roslaunch turtlebot3_gazebo turtlebot3_simulation.launch
============================================================
started roslaunch server http://lau-VirtualBox:42265/

SUMMARY
========

PARAMETERS
 * /cmd_vel_topic_name: /cmd_vel
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /
    turtlebot3_drive (turtlebot3_gazebo/turtlebot3_drive)

ROS_MASTER_URI=http://localhost:11311

process[turtlebot3_drive-1]: started with pid [2677]
[ INFO] [1610002685.141700814]: TurtleBot3 Simulation Node Init
============================================================

after the code, the robot just remain silent there. I am able to control the robot with the teleop_key command.
I suspect is something wrong here roslaunch turtlebot3_gazebo turtlebot3_simulation.launch

ROBOTIS-Will, 2021-01-07:
Sorry but I can't reproduce your problem with my Melodic machine.
I strongly recommend to reinstall Gazebo.
Thanks.

lyhlau, 2021-01-07:
> Sorry but I can't reproduce your problem with my Melodic machine.
> I strongly recommend to reinstall Gazebo.
> Also, it is not recommended to run ROS on virtual machine.
> Thanks.

Thank for your help, your advice is much appreciated.