ROBOTIS-GIT/turtlebot3_simulations #103: Turtlebot Model Waffle Causes Wobble in Cartographer

ClaytonCalabrese, 2020-01-20:
Following the tutorial steps from : http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_simulation/#simulation

>     $ export TURTLEBOT3_MODEL=${TB3_MODEL}
>     $ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
>     $ export TURTLEBOT3_MODEL=${TB3_MODEL}
>     $ ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True

When I use waffle as the ${TB3_MODEL} ,after about 30-45 seconds, the lidar in cartographer would start to spin on the Z axis. Here is an image of the scanners being way off of the level field : https://i.imgur.com/rYw9enr.png .  This wobble continues back and forth and does not stop. It does not matter if the model is moving, the same problem still occurs. It is not running into any objects or terrain either to cause tilt.

The spin causes the robot to lose confidence in the map it has created which causes it to get blurrier as time goes on.

This problem is not encountered when following the exact same steps with burger as the ${TB3_MODEL}

Operating system is Linux Mint 19.02 (an equivalent to ubuntu 18.04).

ROBOTIS-Will, 2020-01-22:
@ClaytonCalabrese 
Thank you for reporting the issue. We'll look into this, however, please note that google cartographer had not been maintained for a while and there could be some functional issue with ROS2.
[https://github.com/googlecartographer/cartographer_ros/issues/1162](https://github.com/googlecartographer/cartographer_ros/issues/1162)

ROBOTIS-Will, 2020-02-28:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if necessary. Thanks.

closed 2020-02-28 by ROBOTIS-Will