ROBOTIS-GIT/turtlebot3_simulations #105: can't locate node [turtlebot3_drive] in package [turtlebot3_gazebo]

CharlieLeee, 2020-02-11:
Hi, thanks for the great work!
When I followed the tutorial: 
http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#turtlebot3-simulation-using-gazebo 
and I tried to run the command: roslaunch turtlebot3_gazebo turtlebot3_simulation.launch
and error occurred:
```
ERROR: cannot launch node of type [turtlebot3_gazebo/turtlebot3_drive]: can't locate node [turtlebot3_drive] in package [turtlebot3_gazebo]
```

ROBOTIS-Will, 2020-02-19:
@CharlieLeee 
Hi,
Please make sure that you have downloaded turtlebot3_simulation meta-package and built it.
```bash
$ cd ~/catkin_ws/src/
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git
$ cd ~/catkin_ws && catkin_make
```

CharlieLeee, 2020-02-19:
@ROBOTIS-Will Hi, thanks for the reply! I have already built turtlebot3_simulations. Is this problem relate to any of the drive issues?  like https://github.com/ROBOTIS-GIT/turtlebot3_simulations/issues/102

ROBOTIS-Will, 2020-02-28:
@CharlieLeee 
Unable to launch a specific node does not have to do with #102 issue.
Your error seems to be related with unsuccessful package installation.

CharlieLeee, 2020-03-01:
@ROBOTIS-Will  Okay, I'll try to rebuild all the packages. Thanks for answering!

closed 2020-03-01 by CharlieLeee