ROBOTIS-GIT/turtlebot3_simulations #107: Adjust covariance matrix

saimouli, 2020-04-18:
Hello, 

I am using ROS Melodic and Waffle model and latest commit (6278566). Repo is up-to-date with the current version. 

rostopic echo /imu gives: 

> header:
seq: 12477
stamp:
secs: 14962
nsecs: 36000000
frame_id: "base_footprint"
orientation:
x: 0.0015566085891899162
y: 0.0003276490868402762
z: -0.9788036429173815
w: 0.2047947709963295
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
x: -0.0005358902335171238
y: -0.0003909577325524005
z: 0.19488579839896925
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration:
x: 5.083136873524716e-05
y: 0.00010048728744395968
z: 2.618836550644121e-07
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

Covariance cannot be zero in the real-world for IMU and needs to reflect that in the simulation. Where can I change the covariance matrix for Imu and odom?
Thanks

ROBOTIS-Will, 2020-04-27:
@saimouli 
According to the sensor_msgs/Imu, the covariance matrix is set to zero when they are unknown.
In order to change the covariance in simulation, you might need to manipulate the imu topic.
In TurtleBot3 Gazebo simulation, imu_plugin is used to simulate the imu sensor.
Thank you.

ashekim, 2020-09-22:
This issue has been closed as there weren't recent activities. 
Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2020-09-22 by ashekim