ROBOTIS-GIT/turtlebot3_simulations #110: No turtlebot topics when running turtlebot3_world.launch

sterlingm, 2020-06-26:
When I run the `turtlebot3_world.launch` file, the only topics listed in `rostopic list` are:

```
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/rosout
/rosout_agg

```

Gazebo starts and I can see the environment and waffle_pi model, but I don't have any topics like cmd_vel, odom, sensor_state, etc. 

The output from running the launch file is:

```
process[gazebo-1]: started with pid [4189]
process[gazebo_gui-2]: started with pid [4194]
process[spawn_urdf-3]: started with pid [4199]
[ INFO] [1593193588.071952227]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1593193588.073046966]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1593193588.115089000]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1593193588.116114211]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1593193588.402999688]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1593193588.432082112, 0.005000000]: Physics dynamic reconfigure ready.
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
[spawn_urdf-3] process has finished cleanly
```

ashekim, 2020-08-18:
Hi. @sterlingm 
I'm sorry for the late reply.

Please test below command?

$ export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:${turtlebot3_gazebo_plugin path}/build
$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:${turtlebot3_gazebo_model path}/models

${turtlebot3_gazebo_plugin_path} should be the actual path to the plugin and model folders, such as $HOME/catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo

Test above and let me know if it still doesn't work,
Thank you.

ashekim, 2020-09-22:
This issue has been closed as there weren't recent activities. 
Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2020-09-22 by ashekim

bkakilli, 2020-10-23:
Hello,

I am having the same problem.

The `GAZEBO_PLUGIN_PATH` instructions in the comment above should work for [11.2.2 Standalone Gazebo Plugin](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#standalone-gazebo-plugin), am I right? However, I am following the instructions for simulation through ROS as instructed here: [11.2.1 ROS packages for Gazebo](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#ros-packages-for-gazebo)

I ran the following commands in order,
1. roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
1. roslaunch turtlebot3_gazebo turtlebot3_world.launch
1. roslaunch turtlebot3_gazebo turtlebot3_simulation.launch

then I can see the gazebo window with `waffle` robot (defined in bashrc). However, the /scan is not being published and robot does not obey teleop commands (they are being published, I manually checked the topic).

I can run TurtleBot3 Simulation using Fake Node without any issue. However Gazebo simulation fails to work. Probably the plugins are not behaving right but I must be missing something. I also checked the answer on issue #117, however gazebo already starts in 'play' mode.

I tried on both Kinetic (Ubuntu 16.04) and Melodic (Ubuntu 18.04), both of them runs on Docker since my host system is Ubuntu 20.04.

Topics:
```bash
> rostopic list
/clock
/cmd_vel
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/gazebo_gui/parameter_descriptions
/gazebo_gui/parameter_updates
/odom
/rosout
/rosout_agg
/scan

> rostopic info /scan 
Type: sensor_msgs/LaserScan

Publishers: None

Subscribers: 
 * /turtlebot3_drive (http://emektar:40609/)

```

A screenshot of all windows:
[image]

Thanks!

bkakilli, 2020-10-24:
I solved my problem by installing full desktop version (`apt install ros-melodic-desktop-full`) in my Dockerfile on top of `ubuntu:18.04` image. Previously I've been using prebuilt `ros:melodic-perception-bionic` image, which fails to run simulation. Hope this helps others!

My reference docker files: https://gist.github.com/bkakilli/50ed23371094e11fa083325a78793e79

andimin182, 2021-05-10:
Hello,
I'm having the same problem with Ubuntu 20.04 and ROS NOETIC. The weird thing is that the first time that i launched the turtlebot3 simulation in empty world the teleop node  worked, but later it stopped working. Now i cannot find the cmd_vel and odom topics anymore once i launch the turtlebot3_empty_world.launch only . After, even though the teleop node is publishing on the cmd_vel topic, there are no subscribers. Someone has any ideas why?
Thanks!!