ROBOTIS-GIT/turtlebot3_simulations #115: ros2 robots hardcoded into the world

joe28965, 2020-07-09:
As the title says, in all the ros2 branches (dashing-devel eloquent-devel ros2-devel and ros2) the turtlebot is placed in the world by having a preset world with the environment (for instance the turtlebot world) and the specific turtlebot model itself (burger/waffle/waffle-pi)

In ros1 the turtlebot model was always spawned in using the gazebo_ros spawner. I was curious what the decision was to not do that in ros2. To me it seems a lot more cluttered right now, each world has 3 different world files (one for each turtelbot model).
If you want to move the robot, you will also need to edit the world file.

To me, this seems like a strange design choice. Is there a reason as to why it's set up like this? I could see if I could change it to use the gazebo_ros spawner next week (for eloquent-devel, that's the branch I use myself) Would people be interested in this change?

ashekim, 2020-08-18:
@joe28965 

I'm sorry for the late reply.
I'll check the issue and get back to you.

Thank you.

shonigmann, 2021-03-04:
On perhaps a related note, I am wondering why there are so many different versions of model descriptions being used... 

- the [world files](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/tree/foxy-devel/turtlebot3_gazebo/worlds) have hard-coded model descriptions as mentioned by @joe28965, 
- the [robot_state_publisher](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/foxy-devel/turtlebot3_gazebo/launch/robot_state_publisher.launch.py) uses the [turtlebot3_description's urdf](https://github.com/ROBOTIS-GIT/turtlebot3/tree/foxy-devel/turtlebot3_description/urdf), 
- the .sdf files in this repository seem unused in the launch files included, but 
- other packages that rely on Turtlebot3, like Navigation2, end up using both [urdf](https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bringup/bringup/launch/tb3_simulation_launch.py) and [sdf](https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bringup/nav2_gazebo_spawner/nav2_gazebo_spawner/nav2_gazebo_spawner.py) robot descriptions. 

I am also curious if I am missing something behind the choice of maintaining so many separate descriptions, or if we can expect these definitions to get merged (e.g. so that they are all built from xacro) in the future? 

Thanks for the insight!

ROBOTIS-Will, 2021-03-04:
Hi @shonigmann 
Thank you for the kind ping on this issue.
Some of those world files are used for machine learning environments and somehow they initially were designed in that way.
I started working on reviewing the turtlebot3 machine learning package, there will be some updates on the simulations package as well to reduce the redundant files in each world directories.
Thank you!

joe28965, 2021-06-09:
Hi @ROBOTIS-Will
I was curious as to what the status of this issue is? I would also be more than happy to make the changes myself, and set it up to use the gazebo_ros spawner. However, I would like to know if these changes would be appreciated, since it will take some time to do

ROBOTIS-Will, 2021-06-10:
Hi @joe28965 
Sorry for taking long for reviewing this. 
Reducing the redundancy by separating world and robot model is desired.
There are developers who want to use turtlebot3_simulations as a standalone package so this will be the direction to move forward.
I'd of course appreciate if you can contribute to clean up this package, but changes should be acceptable for the general users who are using a default TurtleBot3 system to learn ROS.
Thank you!

joe28965, 2021-06-10:
Turns out, not as much work as I had expected.

For both Foxy (#161) and Galactic (#162) I created a pull request. I tested all the worlds, and they seemed to work. I also added the ability to choose where you want to spawn your turtlebot by giving the files a launch parameter (x_pose and y_pose).

I have also left the current world folders intact for both Foxy and Galactic (the folders with the 3 different worlds, one for each turtlebot), since I assumed it would be appreciated for possible compatibility reasons. Although I do think it would be best to remove that for Rolling.

However, I am not quite sure which branch is for Rolling, is that ros2 or ros2-devel? If you could let me know I could also create a pull request for Rolling. Also please let me know if you want me to remove the old world folders for Rolling, as I could do that in one go.

ROBOTIS-Will, 2021-06-10:
Hi @joe28965 
Thanks for your contribution!
You can PR to ros2-devel branch which will be merged into ros2 and applied to relevant branches later.

closed 2021-06-22 by joe28965