ROBOTIS-GIT/turtlebot3_simulations #116: Unable to find uri[model://turtlebot3_waffle]

jessicaleu24, 2020-07-11:
Hi I'm trying to launch the empty world with
$ export TURTLEBOT3_MODEL=waffle
$ ros2 launch turtlebot3_gazebo empty_world.launch.py

But I'm getting the following error:
[gazebo-1] [Wrn] [FuelModelDatabase.cc:248] URI not supported by Fuel [model://turtlebot3_waffle]
[gazebo-1] [Wrn] [SystemPaths.cc:464] File or path does not exist [""] [model://turtlebot3_waffle]
[gazebo-1] Error Code 11 Msg: Unable to find uri[model://turtlebot3_waffle]
[gazebo-1] 

How can I fix it?
(I have Ubuntu 18.04)

squirel2000, 2020-07-24:
Hi I have the same issue, but I use the waffle_pi 
$ export TURTLEBOT3_MODEL=waffle_pi
$ ros2 launch turtlebot3_gazebo empty_world.launch.py

and I got the same error code 
Error Code 11 Msg: Unable to find uri[model://turtlebot3_waffle_pi]
(Ubuntu 18.04)

Thanks

ashekim, 2020-08-17:
Hello @jessicaleu24 , @squirel2000 
I'm sorry for the late reply.

Please refer to the [eManual](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_simulation/#ros-packages-for-gazebo).

Thank you.

ashekim, 2020-09-22:
This issue has been closed as there weren't recent activities. 
Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2020-09-22 by ashekim