ROBOTIS-GIT/turtlebot3_simulations #117: turtlebot3 drive not working

jas705, 2020-08-05:
Hi, I am trying to run a turtlebot3 drive for collision avoidance.
$ export TURTLEBOT3_MODEL=burger
$ roslaunch turtlebot3_gazebo turtlebot3_world.launch
$ roslaunch turtlebot3_gazebo turtlebot3_simulation.launch

when running simulation. launch file the robot should move automatically in the world. In this case, the turtlebot3 drive node is initializing the node but the turtlebot3 is not moving in the gazebo.

ashekim, 2020-08-17:
Hi @jas705 
I'm sorry for the late reply.

Please check [▶] button in Gazebo to start simulation.
If it's playing[▶], please try reinstall package.
Thank you.

ashekim, 2020-09-22:
This issue has been closed as there weren't recent activities. 
Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2020-09-22 by ashekim