ROBOTIS-GIT/turtlebot3_simulations #121: CMake Warning in catkin_make turtlebot3_gazebo

GuancongLuo, 2020-09-16:
My system environment：
Ubuntu 20.04
I using follow command to install gazebo 
`curl -sSL http://get.gazebosim.org | sh
`
`$gazebo --version`
`Gazebo multi-robot simulator, version 11.1.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
`
And them, I install ros_noetic.
Finially, git clone turtlebot3_simulations.

When I catkin make turtlebot3_simulations, I got the CMake Warning as follow:
`CMake Warning at /opt/ros/noetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'gazebo' but neither 'gazebo_INCLUDE_DIRS' nor
  'gazebo_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  turtlebot3_simulations/turtlebot3_gazebo/CMakeLists.txt:40 (catkin_package)
`

GuancongLuo, 2020-09-16:
Maybe is same problem [issue#94](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/issues/94#issue-500578635)

rjshim, 2020-09-16:
@ashekim  can you help him with finding a solution to this issue?  Thanks. Ryan

ashekim, 2020-09-16:
@GuancongLuo 
Hi. Thank you for your questions.
I will check this issue and get back to you.

Thank you.

ashekim, 2020-09-17:
@GuancongLuo 

This issue seems to be caused by problem with the CMakeLists.txt
I looked for issues similar to yours. It would be nice to refer to the [link](https://answers.ros.org/question/253588/ros-kinetic-eigen/). 

Also, I just recommended to ask on the ROS forum.
Please refer to it and let me know the result.
Thank you.

GuancongLuo, 2020-09-18:
using gazebo-11 in the DEPENDS can't resolved the problem!

But it's no matter, gazebo is running normally.

Thank you for you help!

ashekim, 2020-09-21:
@GuancongLuo 
I would greatly appreciate it!
If you have any question or suggest, please ask to me any time. 
Thanks :)

closed 2020-09-21 by ashekim