ROBOTIS-GIT/turtlebot3_simulations #123: env-hooks in ros2 branches

joe28965, 2020-09-18:
I was curious why no [env hooks](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths#env-hooks) are used in any of the ros2 branches. This will eliminate the need for people to manually add the GAZEBO_MODEL_PATH to their bashrc

`echo 'export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models' >> ~/.bashrc`

Is there a specific reason as to why it's preferred to do this manually?

ashekim, 2020-09-21:
@joe28965 

Hi. @joe28965 
We will discuss it in the team and update our package. 
Thank you so much for your contribution.

Rayman, 2020-10-08:
@joe28965 I've created a PR with a much cleaner way here: https://github.com/ROBOTIS-GIT/turtlebot3_simulations/pull/129. I hope you can take a look if it solves your problem.

ROBOTIS-Will, 2021-01-19:
This fix had been merged and released in the recent ROS2 update.
https://github.com/ros/rosdistro/pull/28006
Thank you @Rayman , @joe28965 for your contributions!

PR #129 (2020-12-09): Automatically make the models available to gazebo_ros
By adding gazebo_model_path to the exports of this package, the models
directory will be automatically available if you are using
`gazebo.launch.py`.  This means that you don't need to set
GAZEBO_MODEL_PATH this in your .bashrc or in other places.

This was implemented in the ROS1 branch but the change was
somehow forgotten.

Also fix #123
closed 2021-01-19 by ROBOTIS-Will