ROBOTIS-GIT/turtlebot3_simulations #135: waffle_pi cartographer failing

GrueneNeun, 2020-12-15:
I am running the Turtlebot3 in a gazebo environment with ros foxy on Ubuntu 20.04. While the Burger model is running fine within cartographer as far as i can tell (i let it run for over an hour), the waffle_pi model starts behaving weird some minutes after the start of cartographer:

[image]

I tried this on the ros2, ros2-devel and foxy-devel branch - all waffle_pi models behave the same way.

ROBOTIS-Will, 2020-12-17:
Hi,
It looks like the Rviz has some issue with filtering the IMU sensor data.
Could you turn off the imu data option and see if it fixes the issue?
https://github.com/ROBOTIS-GIT/turtlebot3/blob/foxy-devel/turtlebot3_cartographer/config/turtlebot3_lds_2d.lua#L52

GrueneNeun, 2020-12-18:
Yes, turning off the imu data option clearly helps. But is the IMU sensor data different between the burger and the waffle_pi model?

ROBOTIS-Will, 2020-12-21:
No, both Burger and Waffle Pi use the identical IMU as they are using the IMU on OpenCR1.0 board.

GrueneNeun, 2020-12-21:
But where is the difference then? Burger model is running fine with IMU data option turned on while waffle_pi is not?

ROBOTIS-Will, 2020-12-22:
There was a missing imu joint in the sdf file and this PR(#133) will fix the issue in the gazebo.

GrueneNeun, 2020-12-22:
After seeing that you merged the PR into branch ros2-devel, i switched to this branch, pulled (i did see the change in the 4 sdf files),  turned IMU data back on in the cartographer config and ran my usual setup:

1. ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
2. ros2 run turtlebot3_gazebo turtlebot3_drive
3. ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True

in parallel terminals. While the bot moved very well and avoided objects as intended before the change, it now starts after a short time (usually under one minute) to move backwards, collide with the walls and flip over and over.
If i switch to the burger model in all 3 terminals and run the same commands it still behaves as expected.

ROBOTIS-Will, 2020-12-23:
@GrueneNeun 
Thanks for the update.
I'm currently checking on the Waffle Pi. I'll update the package when the issue is resolved.

GrueneNeun, 2021-01-01:
After this commit the waffle_pi behaves nearly as in the beginning - it avoids objects as expected, but the cartographer does the same flipping as before. Turning off the IMU option in cartographer still helps.

ROBOTIS-Will, 2021-01-04:
@GrueneNeun 
Thank you for the feedback.
I'll run the simulation again and see if I can reproduce the problem.
Happy New Year!

GrueneNeun, 2021-01-06:
It seems i indeed forgot to update the turtlebot3 package - now everything runs as expected.

Thank you for your efforts ;)

closed 2021-01-06 by GrueneNeun