ROBOTIS-GIT/turtlebot3_simulations #136: Possibly wrong collision box for `camera_rgb_optical_frame`

goekce, 2020-12-21:
I simulated Waffle-pi in Gazebo and noticed that there is a collision box defined in front of Waffle-pi:

https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/ff1d521a20c67b516b40361463285872ba14f3dd/turtlebot3_gazebo/models/turtlebot3_waffle_pi/model.sdf#L368-L375

Due to this collision box Waffle-pi crashes early to walls. Is this collision box maybe a forgotten artifact from Intelsense camera or is it a feature?
 
[image]

ROBOTIS-Will, 2020-12-22:
@goekce 
Thank you for reporting the issue :+1: 
I'll remove this collision box in the next patch.

ROBOTIS-Will, 2021-01-19:
Closing this issue as the fix has been applied in the recent release.
Thank you.

closed 2021-01-19 by ROBOTIS-Will