ROBOTIS-GIT/turtlebot3_simulations #139: no rending on gazebo

jwu1980, 2021-01-12:
I am following this instruction to run the turtlebot3 simulation.
https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/

However, I don’t see the rending on all models (burger/waffle/waffle_pi).
Have you seen this before?
[image]

O.S.: Ubuntu 20.04.
ROS: Foxy
Turtlebot3_simulation branch: tried both ros2/foxy-devel

ROBOTIS-Will, 2021-01-13:
@jwu1980 
Hi, the rendering file used to be very large and had too much details in it.
Since simulation does not require such details, there were several requests to simplify the modeling file as much as possible.
If you need more detailed modeling files, you can find them from the previous commit below.
https://github.com/ROBOTIS-GIT/turtlebot3_simulations/tree/58b752a6ac26ae2e3be58b7b0d61af044d570c1f/turtlebot3_gazebo/models
Thank you.

jwu1980, 2021-01-13:
Thanks Willson.

closed 2021-01-13 by jwu1980