ROBOTIS-GIT/turtlebot3_simulations #147: ros2 turtlebot3 ERROR [gzserver-1] process has died

matteoderose, 2021-04-02:
Hi, I'm trying to run turtlebot3 simulations (https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/) and I get this error:

`$ ros2 launch turtlebot3_gazebo empty_world.launch.py`

[INFO] [launch]: All log files can be found below /home/matteodr/.ros/log/2021-04-02-12-48-17-619344-matteodr-11682
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [11684]
[INFO] [gzclient   -2]: process started with pid [11687]
[INFO] [ros2-3]: process started with pid [11690]
[INFO] [robot_state_publisher-4]: process started with pid [11692]
[robot_state_publisher-4] [WARN] [1617360500.000960502] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link base_link had 5 children
[robot_state_publisher-4] Link caster_back_link had 0 children
[robot_state_publisher-4] Link imu_link had 0 children
[robot_state_publisher-4] Link base_scan had 0 children
[robot_state_publisher-4] Link wheel_left_link had 0 children
[robot_state_publisher-4] Link wheel_right_link had 0 children
[robot_state_publisher-4] [INFO] [1617360500.030009294] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-4] [INFO] [1617360500.030081392] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1617360500.030111691] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-4] [INFO] [1617360500.030140416] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-4] [INFO] [1617360500.030170631] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-4] [INFO] [1617360500.030199403] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-4] [INFO] [1617360500.030272511] [robot_state_publisher]: got segment wheel_right_link
[ros2-3] Node not found
[ERROR] [ros2-3]: process has died [pid 11690, exit code 1, cmd 'ros2 param set /gazebo use_sim_time True'].
[ERROR] [gzserver-1]: process has died [pid 11684, exit code 255, cmd 'gzserver /home/matteodr/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/worlds/empty_worlds/burger.model                                                                   -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so       '].

Environment:
- Ubuntu 20.04
- ROS2 foxy
- turtlebot3 installed from source in turtlebot3_ws folder as illustrated on the guide (https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/)

How to solve?

P.S. I can run gazebo alone smoothly.

ROBOTIS-Will, 2021-04-05:
Hi @matteoderose 
Could you try install gazebo packages with the command below?
```
$ sudo apt-get install ros-foxy-gazebo-*
```

matteoderose, 2021-04-05:
Many thanks, problem solved. I reinstalled ros2, gazebo (using your suggestion) and turtlebot3 and now it works.

closed 2021-04-05 by matteoderose