ROBOTIS-GIT/turtlebot3_simulations #149: TB3 slowly drifting in Gazebo

goekce, 2021-04-05:
Using ROS2 foxy and `turtlebot_simulations` v. 2.2.2.

I start the simulation without any controlling node:
```sh
ros2 launch turtlebot3_gazebo turtlebot3_house.launch.py
```
Robot slowly drifts as follows:

[image]

I would expect that the robot would stop when the motors are not driven. I also tried to stop the robot using `teleop_keyboard` after which the robot still drifts.

I tinkered around with wheel friction parameters which did not help, e.g., here:

https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/7603541b97ee824ca41e45c8d2a026eaa36b17e7/turtlebot3_gazebo/models/turtlebot3_waffle_pi/model.sdf#L186-L220

Can someone help me with this?

ROBOTIS-Will, 2021-04-06:
Hi @goekce 
Looks like the inertial values need to be revised.
I'll update these values soon.
Thanks for reporting the issue!

goekce, 2021-04-11:
I recently noticed that the robot flies if I place TB3 Waffle Pi on the empty world in Gazebo. I noticed that the default parameters for the ode solver do not suffice to appropriately simulate TB3 physics. I have to change [ode solver `iters` parameter](http://sdformat.org/spec?ver=1.7&elem=physics#solver_iters) to 150 to avoid flying as Robotis does in its worlds, e.g.,:

https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/77151384a09c25271679d9baec5c5cf6587bd9bd/turtlebot3_gazebo/worlds/empty_worlds/waffle_pi.model#L30-L32

Here is a minimal world with minimal parameters:

```xml
<?xml version='1.0' ?>                                                                                               
<sdf version='1.7'>                                                                                                  
    <world name="world">                                                                                             
        <physics type="ode">                                                                                         
            <ode>                                                                                                    
                <solver>                                                                                             
                    <iters>150</iters>                                                                               
                </solver>                                                                                            
            </ode>                                                                                                   
        </physics>                                                                                                   
        <include><name>sun</name><uri>model://sun</uri></include>                                                    
        <include><name>ground</name><uri>model://ground_plane</uri></include>                                        
        <include><name>tb3</name><uri>model://turtlebot3_waffle_pi</uri><pose>0.5 0.5 0 0 0 1.5707963267948966</pose></include>                                                                                                           
    </world>                                                                                                         
</sdf>
```

If you remove `iters` (then it is set to 50) then the robot flies. Maybe SDFormat default parameters were changed recently. I use Gazebo 11.4.

ROBOTIS-Will, 2021-04-12:
Hi, 
The `iters` tag governs the simulation frequency and higher value uses more resource to create more accurate simulation result.
The drifting is actually caused by incorrect inertia value of left and right wheels, and they can be fixed by modifying `/turtlebot3_simulations/turtlebot3_gazebo/models/turtlebot3_waffle_pi/model.sdf` file as below.

In line 168, where `wheel_left_link` is defined,
```
    <link name="wheel_left_link">

      <inertial>
        <pose>0.0 0.144 0.023 -1.57 0 0</pose>
        <inertia>
          <ixx>1.8158194e-03</ixx>
          <ixy>-9.3392e-12</ixy>
          <ixz>1.04909e-11</ixz>
          <iyy>3.2922126e-03</iyy>
          <iyz>5.75694e-11</iyz>
          <izz>1.8158194e-03</izz>
        </inertia>
        <mass>2.8498940e-02</mass>
      </inertial>
```

In line 226, where `wheel_right_link` is defined,
```
    <link name="wheel_right_link">

      <inertial>
        <pose>0.0 -0.144 0.023 -1.57 0 0</pose>
        <inertia>
          <ixx>1.8158194e-03</ixx>
          <ixy>-9.3392e-12</ixy>
          <ixz>1.04909e-11</ixz>
          <iyy>3.2922126e-03</iyy>
          <iyz>5.75694e-11</iyz>
          <izz>1.8158194e-03</izz>
        </inertia>
        <mass>2.8498940e-02</mass>
      </inertial>
```

goekce, 2021-04-12:
How did you calculate these values Will?

I'll leave the issue here until it is fixed in the codebase.

:heart_decoration: for the quick fix!

ROBOTIS-Will, 2021-04-12:
@goekce 
The inertial value is calculated by the CAD tool based on the 3D modeling file.
I ran a quick test on Gazebo after modifying these values, and the issue was resolved, but I'll check again.
This fix will be updated on `foxy-devel` branch and released in this week.
Thanks!

goekce, 2021-04-14:
I installed the `foxy-devel` branches of `turtlebot3` and `turtlebot3_simulations` branches from scratch, and now the drifting is gone :partying_face: Somehow it was not enough to only patch the TB3 Waffle Pi model.

I also do not need `<iters>150</iters>` anymore. The default setting of sdformat (50) seems to be sufficient. The flying robot was probably also caused by the wrong inertia setting.

closed 2021-04-14 by goekce

Lobotuerk, 2023-09-13:
> @goekce The inertial value is calculated by the CAD tool based on the 3D modeling file. I ran a quick test on Gazebo after modifying these values, and the issue was resolved, but I'll check again. This fix will be updated on `foxy-devel` branch and released in this week. Thanks!

What CAD tool did you use? Did it allow you to set a mass or did it give it to you based on a density of 1? @ROBOTIS-Will