ROBOTIS-GIT/turtlebot3_simulations #152: Waffle keeps spinning in place after starting the world model

nedo99, 2021-04-13:
Hi,

I have an issue when I start turtlebot3 world simulation environment. Turtlebot3 keeps spinning in place forever.

The command I use:
`TURTLEBOT3_MODEL=waffle ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py`

Output:
```
[INFO] [launch]: All log files can be found below /.ros/log/2021-04-13-13-40-15-025665
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_waffle.urdf
/opt/ros/foxy/share/turtlebot3_gazebo/launch/robot_state_publisher.launch.py:47: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/turtlebot3_gazebo/launch/robot_state_publisher.launch.py:47: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [gzserver-1]: process started with pid [1110]
[INFO] [gzclient   -2]: process started with pid [1112]
[INFO] [robot_state_publisher-3]: process started with pid [1115]
[robot_state_publisher-3] [WARN] [1618321215.602239934] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 7 children
[robot_state_publisher-3] Link camera_link had 2 children
[robot_state_publisher-3] Link camera_depth_frame had 1 children
[robot_state_publisher-3] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-3] Link camera_rgb_frame had 1 children
[robot_state_publisher-3] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-3] Link caster_back_left_link had 0 children
[robot_state_publisher-3] Link caster_back_right_link had 0 children
[robot_state_publisher-3] Link imu_link had 0 children
[robot_state_publisher-3] Link base_scan had 0 children
[robot_state_publisher-3] Link wheel_left_link had 0 children
[robot_state_publisher-3] Link wheel_right_link had 0 children
[robot_state_publisher-3] [INFO] [1618321215.604747876] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1618321215.604788132] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1618321215.604797881] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1618321215.604805715] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-3] [INFO] [1618321215.604813626] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-3] [INFO] [1618321215.604821867] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-3] [INFO] [1618321215.604829532] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-3] [INFO] [1618321215.604837264] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-3] [INFO] [1618321215.604845598] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-3] [INFO] [1618321215.604853878] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-3] [INFO] [1618321215.604861406] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1618321215.604868967] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1618321215.604876719] [robot_state_publisher]: got segment wheel_right_link
[gzserver-1] [INFO] [1618321216.478130422] [turtlebot3_imu]: <initial_orientation_as_reference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[gzserver-1] [INFO] [1618321216.874484692] [camera_driver]: Publishing camera info to [/camera/camera_info]
[gzserver-1] [INFO] [1618321216.900551832] [turtlebot3_diff_drive]: Wheel pair 1 separation set to [0.287000m]
[gzserver-1] [INFO] [1618321216.900608397] [turtlebot3_diff_drive]: Wheel pair 1 diameter set to [0.066000m]
[gzserver-1] [INFO] [1618321216.908204340] [turtlebot3_diff_drive]: Subscribed to [/cmd_vel]
[gzserver-1] [INFO] [1618321216.926260959] [turtlebot3_diff_drive]: Advertise odometry on [/odom]
[gzserver-1] [INFO] [1618321216.928142427] [turtlebot3_diff_drive]: Publishing odom transforms between [odom] and [base_footprint]
[gzserver-1] [INFO] [1618321216.939964814] [turtlebot3_joint_state]: Going to publish joint [wheel_left_joint]
[gzserver-1] [INFO] [1618321216.940019103] [turtlebot3_joint_state]: Going to publish joint [wheel_right_joint]
```

My env:
Ubuntu 20.04 and ROS Foxy.

Any hints?

Regards,
Nedim

nedo99, 2021-04-13:
Looking from RVIZ, odom frame makes circles for some reason.

ROBOTIS-Will, 2021-04-14:
Hi,
I wasn't able to reproduce this issue. Could you try with the latest code for Foxy?
You can uninstall any debian packages then download and build the code as shown below.
```bash
$ sudo apt remove ros-foxy-turtlebot3 ros-foxy-turtlebot3-msgs ros-foxy-turtlebot3-simulations
$ cd ~/turtlebot3_ws/src/
$ git clone -b foxy-devel https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
$ git clone -b foxy-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
$ git clone -b foxy-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git
$ cd ~/turtlebot3_ws
$ colcon build --symlink-install
$ source ~/turtlebot3_ws/install/setup.bash
```

nedo99, 2021-04-14:
Did that, but it seems it is not the issue.

I am running it within the docker (with Ubuntu 20.04 and ROS Foxy), but that should not make a difference. Not sure if I am running some prebuilt simulation or something. Is that possible? Is there something else that could affect this? I checked processes, there is not other navigation process or something which sends commands.

nedo99, 2021-04-14:
It seems there was another ros node running which was overlapping with the `odom` frame.

Will close the issue.

Thanks!

closed 2021-04-14 by nedo99