ROBOTIS-GIT/turtlebot3_simulations #154: Spawn

y2d2, 2021-04-21:
Not sure If I should post this here, but i have an issue running the following command on a clean ubuntu 20.04 ros-noetic environment: 
`roslaunch turtlebot3_gazebo turtlebot3_world.launch`

I got following error message : 

`[ERROR] [1619030586.067723, 0.001000]: Spawn service failed. Exiting.
Illegal instruction (core dumped)`
`[gazebo-2] process has died [pid 71, exit code 132, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/yuri/Documents/PhD/ROS_WS/noetic/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_world.world __name:=gazebo __log:=/root/.ros/log/6408ec9a-a2d1-11eb-a88b-6c626d47bfc2/gazebo-2.log].
log file: /root/.ros/log/6408ec9a-a2d1-11eb-a88b-6c626d47bfc2/gazebo-2*.log`
`[spawn_urdf-4] process has died [pid 83, exit code 1, cmd /opt/ros/noetic/lib/gazebo_ros/spawn_model -urdf -model turtlebot3_burger -x -2.0 -y -0.5 -z 0.0 -param robot_description __name:=spawn_urdf __log:=/root/.ros/log/6408ec9a-a2d1-11eb-a88b-6c626d47bfc2/spawn_urdf-4.log].
log file: /root/.ros/log/6408ec9a-a2d1-11eb-a88b-6c626d47bfc2/spawn_urdf-4*.log`

I get this error independent whether I install the packages through apt or if I build the master or noetic-devel branch from this git. 

The spawn_urdf-4*.log mentions this error: 

`...`
`[rospy.internal][INFO] 2021-04-21 18:43:05,023: topic[/clock] adding connection to [http://Ares:39871/], count 0`
`[rospy.internal][WARNING] 2021-04-21 18:43:06,069: Unknown error initiating TCP/IP socket to Ares:42191 (http://Ares:39871/):`
` Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 560, in connect
    self.socket.connect((dest_addr, dest_port))
ConnectionRefusedError: [Errno 111] Connection refused`

`[rospy.internal][INFO] 2021-04-21 18:43:06,070: topic[/clock] removing connection to http://Ares:39871/`

I tried the same with a clean ubuntu 18.04 melodic environment and there it works fine.  
I also tried a similar simulation with a different robot in the ubuntu 20.04 ros-noetic environment and it works fine as well. 

Maybe I overlooked something simple ? 

Cheers, 
y2d2

ROBOTIS-Will, 2021-04-22:
Hi @y2d2 
Just fresh installed Noetic on Ubuntu 20.04 and compiled the latest source code with `git clone -b noetic-devel ...`, but wasn't able to reproduce the issue.
Have you tried reinstalling the Gazebo?
```bash
$ sudo apt-get install ros-noetic-gazebo-ros*
```
```
... logging to /home/robotis/.ros/log/b5f1019c-a311-11eb-842e-39de105b558e/roslaunch-robotis-NUC8i5BEK-33425.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://localhost:38913/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [33471]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b5f1019c-a311-11eb-842e-39de105b558e
process[rosout-1]: started with pid [33489]
started core service [/rosout]
process[gazebo-2]: started with pid [33492]
process[gazebo_gui-3]: started with pid [33496]
process[spawn_urdf-4]: started with pid [33501]
[ INFO] [1619058204.483887967]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1619058204.485223575]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1619058204.619189320]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1619058204.620027894]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1619058204.999434260]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1619058205.024724246, 0.006000000]: Physics dynamic reconfigure ready.
[ INFO] [1619058205.370855180, 0.141000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1619058205.370924394, 0.141000000]: Starting Laser Plugin (ns = /)
[ INFO] [1619058205.372272678, 0.141000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1619058205.466480901, 0.141000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1619058205.466556532, 0.141000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1619058205.467294659, 0.141000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1619058205.468179779, 0.141000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1619058205.468795154, 0.141000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1619058205.470855935, 0.141000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1619058205.471244257, 0.141000000]: DiffDrive(ns = //): Advertise odom on odom 
[spawn_urdf-4] process has finished cleanly
log file: /home/robotis/.ros/log/b5f1019c-a311-11eb-842e-39de105b558e/spawn_urdf-4*.log
```

y2d2, 2021-04-22:
hi @ROBOTIS-Will , 

Thanks for trying to reproduce the error. 
I tried what you suggested 

> $ sudo apt-get install ros-noetic-gazebo-ros*

And it installed some additional libraries, but unfortunately the error remained. (I'm also not convinced it has to do with Gazebo since the other robot simulation does not have this particular error.) 

Anyway I proceeded to remove gazebo 
`apt-get purge ros-noetic-gazebo*`
`apt-get autoremove`

and reinstalled gazebo 
`apt-get install ros-noetic-gazebo*`

Had to reinstall xacro as well 
`apt-get install ros-noetic-xacro`

Yet the issue still remains.... 

I'm quite sure i have a clean and fresh system since I use a docker, which its only purpose is to run simulations.
(Of course a run the container with --privileged and other options so the container can access the host display and other driver files etc...) and as mentioned before the setup works fine with another robot in ubuntu 20.04 ros noetic and with the turtlebot in ubuntu 18.04 and ros melodic. 

Maybe it has to do with the .urdf file ? I noticed they are part of the turtlebot3 package and not the turtlebot3-simulation package. Which branch should I use for this git? ( I tried the main and the noetic-devel. same error though so not sure it is this neither.)

ROBOTIS-Will, 2021-04-22:
Oh, sorry about that. I mistakenly assumed that you have follows all installation steps in the eManual.
The URDF is included in the turtlebot3 package and you can either install from binary or build from the source.
```
$ sudo apt install ros-noetic-turtlebot3
```
or
```
$ cd ~/catkin_ws/src
$ git clone -b noetic-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
```
Thank you.

y2d2, 2021-04-26:
So just to be clear as mentioned in the previous communication I did clone this git (git clone -b noetic-devel https://github.com/ROBOTIS-GIT/turtlebot3.git) into my workspace. I tried with both the master branch and the noetic-devel branch. 

I also tried to install them directly through apt. In all 3 case I have the same failure. 

I did not know about the eManual. 
-> I restarted followed the steps there (except installed ROS1 noetic through the official installation guide and not the script.)
-> Build from source with noetic-devel branch for 4 gits: 
   - git clone -b noetic-devel https://github.com/ROBOTIS-GIT/DynamixelSDK.git
   - git clone -b noetic-devel https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
   - git clone -b noetic-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
   -  git clone -b noetic-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git

Used catkin_make to build without error sourced the catkin_ws (./devel/setup.bash)
Run roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch

Same error... 

Not sure what is going wrong.
Used rosdep in the catkin_ws to check if I was missing something, but it returned: 
rosdep install -r --from-paths . --ignore-src --rosdistro noetic -y
#All required rosdeps installed successfully

Cheers, 

Yuri

ROBOTIS-Will, 2021-04-27:
@y2d2 
Hmm... I don't see anything unusual in your installation process.
Are you running ROS on a virtual machine by any chance?

y2d2, 2021-04-27:
Yes it is running in a docker container. 
Though it seems strange that the same setup with ubuntu 18.04 and melodic works fine in a docker container. 
And another robot is working as well in this  ubuntu 20.04 noetic docker container. 

I tried now to separate the commands: 
First launched an empty world: 
`roslaunch gazebo_ros empty_world.launch`
then spawned the model with: 
`rosrun gazebo_ros spawn_model -file $rosws/src/turtlebot3/turtlebot3_description/urdf/turtlebot3_burger.gazebo.xacro -urdf -x 0 -y 0 -z 0 -model burger`
The last command gives back the earlier spawn error. When I look into the terminal where gazebo is running I get this error message: 
`Error [parser_urdf.cc:3193] Unable to call parseURDF on robot model`
`Error [parser.cc:488] parse as old deprecated model file failed.`

Maybe this might give some more clues on what might go wrong?

Just to be sure here are some of the version of the setup: 
____Ubuntu____
`$ lsb_release -a`
`LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch`
`Distributor ID:	Ubuntu`
`Description:	Ubuntu 20.04.2 LTS`
`Release:	20.04`
`Codename:	focal`

____ros_____
`$ rosversion -d`
`noetic`

____Gazebo_____
`$ gazebo --version`
`Gazebo multi-robot simulator, version 11.3.0`
`Copyright (C) 2012 Open Source Robotics Foundation.`
`Released under the Apache 2 License.`
`http://gazebosim.org`

_____Ros_gazebo___
`$ rosversion gazebo_ros`
`2.9.1`

Cheers and thanks al ready for the effort !

Yuri

ROBOTIS-Will, 2021-04-28:
@y2d2 
Unfortunately, I don't have much information about running a simulation on a docker environment, I believe there will be much more useful resources in [ROS Answers](https://answers.ros.org/questions/).
Regarding the URDF parsing error when loading the TurtleBot3 file, you should load the urdf.xacro instead of gazebo.xacro. gazebo.xacro does not contain any urdf information.
Thanks!

kimkimyoung, 2021-09-16:
[image]
Just change the order of sys.path, make sure python2 is in the last position. This solution solved my problem.

JiangxinkeTao, 2023-04-11:
I have the same problem, did you solve the problem？
there are  my logs:
roslaunch gazebo_ros empty_world.launch 
... logging to /home/tomas/.ros/log/1d190a30-d84f-11ed-b4a2-7cd30a92f045/roslaunch-tomas-3973.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:42129/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [3994]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1d190a30-d84f-11ed-b4a2-7cd30a92f045
process[rosout-1]: started with pid [4012]
started core service [/rosout]
process[gazebo-2]: started with pid [4015]
process[gazebo_gui-3]: started with pid [4022]
[ INFO] [1681206983.286601634]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1681206983.287766368]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1681206983.376635989]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1681206983.377788911]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1681206983.534538307]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1681206983.555560083, 0.006000000]: Physics dynamic reconfigure ready.
Illegal instruction (core dumped)
[gazebo_gui-3] process has died [pid 4022, exit code 132, cmd /opt/ros/melodic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/tomas/.ros/log/1d190a30-d84f-11ed-b4a2-7cd30a92f045/gazebo_gui-3.log].
log file: /home/tomas/.ros/log/1d190a30-d84f-11ed-b4a2-7cd30a92f045/gazebo_gui-3*.log

xiaoxueshengyao, 2024-09-05:
Same problem again. It happened as @y2d2 saied. Nomatter installed in apt or make from source.

xiaoxueshengyao, 2024-09-06:
> Same problem again. It happened as @y2d2 said. No matter installed in apt or make from source. And the tf tree only have odom and base_footprint.

I thought the node will publish the tf before, while it's my fault. I launch the state publish node so I can get the tf, and the spawn node crashed didn't seem to have any effect. The spawn node just died after send the urdf to gazebo, is it right?

sunghowoo, 2025-02-04:
The original post is too old, so I will close it. If there is an issue, please reopen it with the latest Noetic or Humble version.

closed 2025-02-04 by sunghowoo