ROBOTIS-GIT/turtlebot3_simulations #155: Unable get amcl_pose datas when I wanted to simulate with multirobot.

AaronWaxson, 2021-04-24:
I want to simulate with multi turtlebot3s to check my algorithm. But when I wanted to launch multi turtlebot3 with amcl, I failed. I couldn't get lasers' datas. 

"multi_turtlebot3.launch"
`
<launch>
  <arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
  <arg name="first_tb3"  default="tb3_0"/>
  <arg name="second_tb3" default="tb3_1"/>

  <arg name="first_tb3_x_pos" default=" -2.0"/>
  <arg name="first_tb3_y_pos" default=" -0.5"/>
  <arg name="first_tb3_z_pos" default=" 0.0"/>
  <arg name="first_tb3_yaw"   default=" 0.0"/>

  <arg name="second_tb3_x_pos" default=" 2.0"/>
  <arg name="second_tb3_y_pos" default=" -0.5"/>
  <arg name="second_tb3_z_pos" default=" 0.0"/>
  <arg name="second_tb3_yaw"   default=" 0.0"/>

  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find turtlebot3_gazebo)/worlds/turtlebot3_world.world"/>
    <arg name="paused" value="false"/>
    <arg name="use_sim_time" value="true"/>
    <arg name="gui" value="true"/>
    <arg name="headless" value="false"/>
    <arg name="debug" value="false"/>
</include>  

  <group ns = "$(arg first_tb3)">   
    <include file="$(find turtlebot3_bringup)/launch/turtlebot3_remote.launch">
      <arg name="model" value="$(arg model)" />
      <param name="tf_prefix" value="$(arg first_tb3)" />
    </include>

    <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-urdf -model $(arg first_tb3) -x $(arg first_tb3_x_pos) -y $(arg first_tb3_y_pos) -z $(arg first_tb3_z_pos) -Y $(arg first_tb3_yaw) -param robot_description" />
    
    <include file="$(find diff_wheel_robot_nav)/launch/my_amcl.launch">
      <arg name="ns"     default="$(arg first_tb3)"/>
      <arg name="initial_pose_x" default="-2.0"/>
      <arg name="initial_pose_y" default="-0.5"/>
      <arg name="initial_pose_a" default="0.0"/>
    </include>

      
    <node pkg="map_server" type="map_server" name="map_server" 
      args="$(find diff_wheel_robot_nav)/maps/mapWithoutBound.yaml"/>
  </group>

  <group ns = "$(arg second_tb3)">   
    <include file="$(find turtlebot3_bringup)/launch/turtlebot3_remote.launch">
      <arg name="model" value="$(arg model)" />
      <param name="tf_prefix" value="$(arg second_tb3)" />
    </include>

    <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-urdf -model $(arg second_tb3) -x $(arg second_tb3_x_pos) -y $(arg second_tb3_y_pos) -z $(arg second_tb3_z_pos) -Y $(arg second_tb3_yaw) -param robot_description" />
    
    <include file="$(find diff_wheel_robot_nav)/launch/my_amcl.launch">
      <arg name="ns"     default="$(arg second_tb3)"/>
      <arg name="initial_pose_x" default="-2.0"/>
      <arg name="initial_pose_y" default="-0.5"/>
      <arg name="initial_pose_a" default="0.0"/>
    </include>

      
    <node pkg="map_server" type="map_server" name="map_server" 
      args="$(find diff_wheel_robot_nav)/maps/mapWithoutBound.yaml"/>
  </group>

  
	<node pkg="rviz" type="rviz" name="rviz" required="true"
		args="-d $(find diff_wheel_robot_nav)/rviz/my_multi_turtlebot3.rviz"/>
</launch>
`
"my_amcl.launch"
`
<launch>
	
	<arg name="ns" default=""/>
	<arg name="scan_topic"     default="scan"/>
	<arg name="initial_pose_x" default="-2.0"/>
	<arg name="initial_pose_y" default="-0.5"/>
	<arg name="initial_pose_a" default="0.0"/>

	
	<node pkg="amcl" type="amcl" name="amcl">
		<param name="min_particles"             value="500"/>
		<param name="max_particles"             value="3000"/>
		<param name="kld_err"                   value="0.02"/>
		<param name="update_min_d"              value="0.20"/>
		<param name="update_min_a"              value="0.20"/>
		<param name="resample_interval"         value="1"/>
		<param name="transform_tolerance"       value="0.5"/>
		<param name="recovery_alpha_slow"       value="0.00"/>
		<param name="recovery_alpha_fast"       value="0.00"/>
		<param name="initial_pose_x"            value="$(arg initial_pose_x)"/>
		<param name="initial_pose_y"            value="$(arg initial_pose_y)"/>
		<param name="initial_pose_a"            value="$(arg initial_pose_a)"/>
		<param name="gui_publish_rate"          value="50.0"/>

		<remap from="scan"                      to="scan"/>
		<param name="laser_max_range"           value="3.5"/>
		<param name="laser_max_beams"           value="180"/>
		<param name="laser_z_hit"               value="0.5"/>
		<param name="laser_z_short"             value="0.05"/>
		<param name="laser_z_max"               value="0.05"/>
		<param name="laser_z_rand"              value="0.5"/>
		<param name="laser_sigma_hit"           value="0.2"/>
		<param name="laser_lambda_short"        value="0.1"/>
		<param name="laser_likelihood_max_dist" value="2.0"/>
		<param name="laser_model_type"          value="likelihood_field"/>

		<param name="odom_model_type"           value="diff"/>
		<param name="odom_alpha1"               value="0.1"/>
		<param name="odom_alpha2"               value="0.1"/>
		<param name="odom_alpha3"               value="0.1"/>
		<param name="odom_alpha4"               value="0.1"/>

		<param name="odom_frame_id"             value="$(arg ns)/odom"/>
		<param name="base_frame_id"             value="$(arg ns)/base_footprint"/>

	</node>
 </launch>
`

When I launch the file, I got the message like these:

log file: /home/aaron/.ros/log/427a48f2-a4da-11eb-91c3-65637e2cf421/tb3_1-spawn_urdf-9*.log
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame wheel_left_link at time 0.660000 according to authority unknown_publisher at line [ WARN] [1619254292.831645910, 0.661000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame wheel_left_link at time 0.660000 according to authority unknown_publisher 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
[ WARN] [1619254293.285058609, 1.094000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame wheel_left_link at time 1.093000 according to authority unknown_publisher

How should I do to can get the robots' sensors' datas right? Is tf's problem? I searched a lot in the internet but still couldn't find a way to fix that, please help. THANKS!!!!!

ROBOTIS-Will, 2021-06-14:
Hi @AaronWaxson 
I'm sorry about delayed response on this.
There's an example of loading multiple TurtleBot3 for SLAM as below.
Have you tried the example?
https://emanual.robotis.com/docs/en/platform/turtlebot3/basic_examples/#load-multiple-turtlebot3s

When loading multiple robots, each robot will have identical TF, topics, nodes, etc, therefore, you should separate them with different namespaces.

sunghowoo, 2025-02-04:
This issue has been inactive for a long time, so I will close it. If there is a problem, please reopen it with the latest version.

closed 2025-02-04 by sunghowoo