ROBOTIS-GIT/turtlebot3_simulations #159: Release for ROS2 Galactic

SteveMacenski, 2021-06-02:


ROBOTIS-Will, 2021-06-04:
The binary for galactic is being prepared for the release in next week.
Thank you.

ROBOTIS-Will, 2021-06-14:
https://github.com/ros/rosdistro/pull/29888

marcomasa, 2021-07-28:
I recently switched to galactic and tried to use the [turtlebot3 simulation](https://navigation.ros.org/getting_started/index.html) .

However, Gazebo will not start because it cannot find the missing turtlebot3 models.
This is because they cannot be installed via apt at the moment?

`sudo apt install ros-galactic-turtle
ros-galactic-turtlebot3-msgs         ros-galactic-turtlebot3-msgs-dbgsym  ros-galactic-turtlesim               ros-galactic-turtlesim-dbgsym   `

The package that would be required (turtlebot3_gazebo) is missing here.
Do i need to update my sources or something?
I followed the galactic install instructions [here](https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html#id7)

I also cannot make it work by downloading from this repo.
I switched to the ROS2 branch and did a colcon build, added the path to the package 
to the GAZEBO_MODEL_PATH, but then the package structure is not ok?

`evocortex@sim2:~$ gazebo --verbose dev_ws/src/evo_ros2_sim_bringup/worlds/waffle.model 
Gazebo multi-robot simulator, version 11.7.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.7.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.7.21
[Wrn] [ModelDatabase.cc:340] Getting models from[http://models.gazebosim.org/]. This may take a few seconds.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.7.21
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_world]
Error Code 12 Msg: Unable to find uri[model://turtlebot3_world]
[Msg] Loading world file [/home/evocortex/dev_ws/src/evo_ros2_sim_bringup/worlds/waffle.model]
[Err] [Joint.cc:297] EXCEPTION: Couldn't Find Child Link[camera_rgb_frame]

[Err] [Model.cc:272] LoadJoint Failed
[INFO] [1627462668.927258200] [gazebo_ros_node]: ROS was initialized without arguments.
1627462668.928293 [0]   gzserver: using network interface enp5s0 (udp/10.0.7.21) selected arbitrarily from: enp5s0, docker0
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/evocortex/dev_ws/src/turtlebot3_simulations/turtlebot3_gazebo/include"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/evocortex/dev_ws/src/turtlebot3_simulations/turtlebot3_gazebo/launch"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/evocortex/dev_ws/src/turtlebot3_simulations/turtlebot3_gazebo/models"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/evocortex/dev_ws/src/turtlebot3_simulations/turtlebot3_gazebo/rviz"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/evocortex/dev_ws/src/turtlebot3_simulations/turtlebot3_gazebo/src"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/evocortex/dev_ws/src/turtlebot3_simulations/turtlebot3_gazebo/urdf"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/evocortex/dev_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds"
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[INFO] [1627462669.085127216] [intel_realsense_r200_depth_driver]: Publishing camera info to [/intel_realsense_r200_depth/camera_info]
[INFO] [1627462669.085613632] [intel_realsense_r200_depth_driver]: Publishing depth camera info to [/intel_realsense_r200_depth/depth/camera_info]
[INFO] [1627462669.085804351] [intel_realsense_r200_depth_driver]: Publishing pointcloud to [/intel_realsense_r200_depth/points]
[INFO] [1627462669.104149322] [turtlebot3_diff_drive]: Wheel pair 1 separation set to [0.287000m]
[INFO] [1627462669.104192663] [turtlebot3_diff_drive]: Wheel pair 1 diameter set to [0.066000m]
[INFO] [1627462669.104637000] [turtlebot3_diff_drive]: Subscribed to [/cmd_vel]
[INFO] [1627462669.105443719] [turtlebot3_diff_drive]: Advertise odometry on [/odom]
[INFO] [1627462669.106281036] [turtlebot3_diff_drive]: Publishing odom transforms between [odom] and [base_footprint]
[INFO] [1627462669.113339019] [turtlebot3_joint_state]: Going to publish joint [wheel_left_joint]
[INFO] [1627462669.113386839] [turtlebot3_joint_state]: Going to publish joint [wheel_right_joint]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/waffle_base.dae]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/waffle_base.dae]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/waffle_base.dae]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/lds.dae]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/lds.dae]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/lds.dae]
[... 8 lines cut ...]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/tire.dae]
[... cut ...]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/tire.dae]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/tire.dae]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/tire.dae]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://turtlebot3_waffle/meshes/tire.dae]
`

ROBOTIS-Will, 2021-07-29:
Hi @marcomasa 

TurtleBot3 packages for Galactic is waiting for the binary release so at the moment, you cannot install with apt install command.
http://repo.ros2.org/status_page/ros_galactic_default.html?q=turtlebot3

Download the galactic source in the src directory of your workspace and build.
```bash
$ git clone -b galactic-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git
```

The turtlebot3_simulation package was updated to share the common mesh files in order to reduce the size of the package, I'll reflect this change into the model file.
But, for now, you can replace all `model://turtlebot3_waffle/meshes...` to `model://turtlebot3_common/meshes...` in the `waffle.model` file.

Thank you.

GyuH13, 2025-02-04:
As this issue is considered resolved, we will close it. If you have any further questions, please feel free to open a new issue.

closed 2025-02-04 by GyuH13