ROBOTIS-GIT/turtlebot3_simulations #164: Multi-turtlebot3 launch

rllyryan, 2021-06-21:
Will there be a launch file for multiple turtlebot3 launches for foxy-devel?

I am aware that ROS2 launch files have a slightly different syntax than in ROS1 (i.e. Melodic)

Are there any resources for me to learn how to write a multiple robot launch in ROS2?

Thanks in advance!

ROBOTIS-Will, 2021-06-23:
Hi @rllyryan 
I'll consider to add such feature, but please allow us some time to assign time slot for it.
It is also welcomed to contribute your multi robot launch file.
Thank you.

GyuH13, 2025-02-04:
Hi @rllyryan 
Sorry for the late response.
We will soon provide a launch file for multiple TB3 robots in Humble.
Closing this issue now. If you need any further assistance, feel free to open a new issue anytime!

closed 2025-02-04 by GyuH13