ROBOTIS-GIT/turtlebot3_simulations #169: how to use  multi_turtlebot3_slam.launch

SmileCaoJ, 2021-07-16:
i want to tun  multi turtlebot3 slam
can you provide commands to run ：
multi_turtlebot3.launch
multi_turtlebot3_slam.launch

ROBOTIS-Will, 2021-07-19:
Hi, @SmileCaoJ 
You can find more information about multi turtlebot slam from the eManual.
https://emanual.robotis.com/docs/en/platform/turtlebot3/slam_simulation/

Detailed instructions can be unfolded at the end of the page as shown below.
[image]

Thanks!

zorawar12, 2021-11-05:
Hello, The link which you have provided is for ROS1, can you please provide more details for ROS2 Foxy?

zorawar12, 2021-11-05:
I was trying to create a multi-robot publisher subscriber, but I am stuck at this point:

[INFO] [launch]: All log files can be found below /root/.ros/log/2021-11-05-04-55-13-178914-43f72736656e-60779
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [60781]
[INFO] [gzclient   -2]: process started with pid [60784]
[INFO] [ros2-3]: process started with pid [60787]
[INFO] [spawn_tb3-4]: process started with pid [60789]
[INFO] [spawn_tb3-5]: process started with pid [60791]
[INFO] [spawn_tb3-6]: process started with pid [60793]
[INFO] [spawn_tb3-7]: process started with pid [60795]
[INFO] [spawn_tb3-8]: process started with pid [60797]
[INFO] [spawn_tb3-9]: process started with pid [60799]
[INFO] [spawn_tb3-10]: process started with pid [60801]
[INFO] [spawn_tb3-11]: process started with pid [60803]
[INFO] [spawn_tb3-12]: process started with pid [60805]
[INFO] [spawn_tb3-13]: process started with pid [60808]
[ERROR] [gzserver-1]: process has died [pid 60781, exit code 255, cmd 'gzserver                                                                       -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[spawn_tb3-6] [INFO] [1636088114.631804433] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-4] [INFO] [1636088114.633664358] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-8] [INFO] [1636088114.679993802] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-11] [INFO] [1636088114.721817091] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-7] [INFO] [1636088114.731126055] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-9] [INFO] [1636088114.732666897] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-10] [INFO] [1636088114.736860581] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-5] [INFO] [1636088114.753322232] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-13] [INFO] [1636088114.788017188] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-12] [INFO] [1636088114.830102478] [entity_spawner]: Creating Service client to connect to `/spawn_entity`
[spawn_tb3-4] [INFO] [1636088114.838105174] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-6] [INFO] [1636088114.845386148] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-6] [INFO] [1636088114.845848637] [entity_spawner]: Done! Shutting down node.
[spawn_tb3-4] [INFO] [1636088114.851130946] [entity_spawner]: Done! Shutting down node.
[spawn_tb3-6] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[spawn_tb3-9] [INFO] [1636088114.929537513] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-9] [INFO] [1636088114.930014176] [entity_spawner]: Done! Shutting down node.
[spawn_tb3-4] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[spawn_tb3-8] [INFO] [1636088114.960174290] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-8] [INFO] [1636088114.960661618] [entity_spawner]: Done! Shutting down node.
[spawn_tb3-11] [INFO] [1636088114.967483431] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-11] [INFO] [1636088114.967972837] [entity_spawner]: Done! Shutting down node.
[INFO] [spawn_tb3-6]: process has finished cleanly [pid 60793]
[spawn_tb3-7] [INFO] [1636088114.984250193] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-7] [INFO] [1636088114.985397108] [entity_spawner]: Done! Shutting down node.
[spawn_tb3-13] [INFO] [1636088114.985569411] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-13] [INFO] [1636088114.986042042] [entity_spawner]: Done! Shutting down node.
[spawn_tb3-8] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[spawn_tb3-10] [INFO] [1636088115.011123227] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-10] [INFO] [1636088115.011606507] [entity_spawner]: Done! Shutting down node.
[spawn_tb3-9] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[spawn_tb3-7] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[INFO] [spawn_tb3-4]: process has finished cleanly [pid 60789]
[spawn_tb3-11] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[spawn_tb3-13] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[spawn_tb3-12] [INFO] [1636088115.039478503] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-12] [INFO] [1636088115.039978211] [entity_spawner]: Done! Shutting down node.
[spawn_tb3-10] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[INFO] [spawn_tb3-8]: process has finished cleanly [pid 60797]
[INFO] [spawn_tb3-10]: process has finished cleanly [pid 60801]
[INFO] [spawn_tb3-9]: process has finished cleanly [pid 60799]
[INFO] [spawn_tb3-7]: process has finished cleanly [pid 60795]
[INFO] [spawn_tb3-11]: process has finished cleanly [pid 60803]
[spawn_tb3-12] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[INFO] [spawn_tb3-13]: process has finished cleanly [pid 60808]
[spawn_tb3-5] [INFO] [1636088115.133746879] [entity_spawner]: Connecting to `/spawn_entity` service...
[spawn_tb3-5] [INFO] [1636088115.134218905] [entity_spawner]: Done! Shutting down node.
[INFO] [spawn_tb3-12]: process has finished cleanly [pid 60805]
[spawn_tb3-5] /opt/ros/foxy/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[ros2-3] Set parameter successful
[INFO] [spawn_tb3-5]: process has finished cleanly [pid 60791]
[INFO] [ros2-3]: process has finished cleanly [pid 60787]
[INFO] [gzclient   -2]: process has finished cleanly [pid 60784]

sunghowoo, 2025-02-04:
The multi-robot-related update is scheduled to be released in the Humble version this week. Since the original post is too old, I will close this issue.

closed 2025-02-04 by sunghowoo