ROBOTIS-GIT/turtlebot3_simulations #170: Inconsistent torque setting in gazebo simulation

squizz617, 2021-09-01:
Hi. When given the same Twist message via `/cmd_vel` topic, the actual robot and the gazebo simulation show considerable discrepancy in their movements. Would you take a look at my analysis below and share some thoughts?

1. One of the straightforward reasons seem to be the `max_wheel_torque` parameter being set to 20 in the [SDF model](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/foxy-devel/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf#L381-L383), which is way larger than the actual operating torque of the Dynamixel motors. The same applies to the Waffle. Because of this, the simulated turtlebot acc/decelerates at a much faster rate than physically possible.

2. Another source of discrepancy comes from the fact that gazebo simulation relies entirely on the `gazebo_ros_diff_drive` plugin for moving the turtlebot, and the plugin is agnostic of the maximum velocity settings of the physical motors. So if the Twist message sets the target velocity to something greater than the physical maximum (e.g., linear velocity > 0.22), the simulation keeps accelerating towards that target. I've looked into the code to see if the linear/angular velocity can be gracefully capped within the plugin, but could't find any.

ashekim, 2021-11-17:
Hi @squizz617. 
Thanks for your interest in TurtleBot3 and sorry for the late reply.
We will check the issue and reply to you.
thank you for your patience

sunghowoo, 2025-02-04:
We have resumed support for the open platform, but this issue is too old. Currently, TurtleBot will only support Noetic and Humble. If you try again and the issue persists, please feel free to reopen it. Thank you for your understanding.

closed 2025-02-04 by sunghowoo