ROBOTIS-GIT/turtlebot3_simulations #174: Unable catkin_make when I wanted to simulate with turtlebot3.

nfhe, 2021-11-16:
I want to simulate with  turtlebot3 to check my algorithm.I downloaded the turtlebot3_simulations  and catkin_ make, but  there is  a problem, can you help me solve it? Thanks a lot.

`
CMake Error at /home/he/miniconda3/envs/torch-he/lib/python3.7/site-packages/cmake/data/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Protobuf (missing: Protobuf_INCLUDE_DIR)
Call Stack (most recent call first):
  /home/he/miniconda3/envs/torch-he/lib/python3.7/site-packages/cmake/data/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /home/he/miniconda3/envs/torch-he/lib/python3.7/site-packages/cmake/data/share/cmake-3.21/Modules/FindProtobuf.cmake:650 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  robot_track/turtlebot3_simulations/turtlebot3_gazebo/CMakeLists.txt:24 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/he/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/he/catkin_ws/build/CMakeFiles/CMakeError.log".
`

ashekim, 2021-11-17:
Hi @nfhe. 
It seems to be an error that occurs because the Protobuf compiler is not installed.
Please refer to this [link,](https://github.com/Tencent/ncnn/issues/1342) I think it will help you with the solution
Thank you.

nfhe, 2021-11-22:
Hi,@ashekim.

I have solved this problem, but when I load the gazebo environment.
roslaunch turbo 3_ gazebo turtlebot3_ empty_ World. Launch.
 There is only environment, but there is no robot model. I want to ask, how should this problem be solved? thank you very much

(rostorch) he@he-Lenovo-Legion-R7000P2020H:~/catkin_ws$ roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
... logging to /home/he/.ros/log/159b3802-4b95-11ec-830b-8c8caa32777f/roslaunch-he-Lenovo-Legion-R7000P2020H-17510.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://he-Lenovo-Legion-R7000P2020H:33325/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [17525]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 159b3802-4b95-11ec-830b-8c8caa32777f
process[rosout-1]: started with pid [17536]
started core service [/rosout]
process[gazebo-2]: started with pid [17542]
process[gazebo_gui-3]: started with pid [17548]
process[spawn_urdf-4]: started with pid [17552]

ashekim, 2021-11-23:
Hi again.
To run turtlebot3 gazebo, you need to install additional packages besides turtlebot3_simulation.
Please refer to the [link](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#install-turtlebot3-packages-1) to install the package. Then run gazebo.
Thank you.

nfhe, 2021-11-23:
Hi, @ashekim 
thank you for your reply.
I may not have described the problem clearly.

First of all thank you for integrating ROS with Conda, ROS is noetic.
I have followed your instruction to setup Conda env, which called "rostorch". However, here I face some problems when
I use TurtleBot3 but Gazebo can't display the robot.The details are given below.

**Problem Description**:
I load the filehttps://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#gazebo-simulation.I have turtlebot3,turtlebot3_msgs and turtlebot3_simulations in catkin_ws.When I start the gazebo simulation using the tutorial, there are no error warnings, but there is no robot model in the gazebo environment

**Here are my detailed operations**
1.conda activate rostorch
2.cd catkin_ws
3.catkin_make
4.source devel/setup.bash
5.export TURTLEBOT3_MODEL=waffle
6.roslaunch turtlebot3_gazebo turtlebot3_world.launch

**Terminal display**
... logging to /home/he/.ros/log/42dd6c9c-4c0b-11ec-bfb7-8c8caa32777f/roslaunch-he-Lenovo-Legion-R7000P2020H-12831.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://he-Lenovo-Legion-R7000P2020H:43655/

SUMMARY
PARAMETERS

/gazebo/enable_ros_network: True
/robot_description: <?xml version="1....
/rosdistro: noetic
/rosversion: 1.15.9
/use_sim_time: True
NODES
/
gazebo (gazebo_ros/gzserver)
gazebo_gui (gazebo_ros/gzclient)
spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [12844]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 42dd6c9c-4c0b-11ec-bfb7-8c8caa32777f
process[rosout-1]: started with pid [12855]
started core service [/rosout]
process[gazebo-2]: started with pid [12859]
process[gazebo_gui-3]: started with pid [12865]
process[spawn_urdf-4]: started with pid [12871].

**rostopic list**

when I use roslaunch turtlebot3_gazebo turtlebot3_world.launch,the rostopic list only has / clock / rosout / rosout_ agg.

**End**

I found two and didn't know where the problem was. I look forward to your reply.

Thank you

ashekim, 2021-11-23:
Hi. @nfhe.
i See.
When I entered this command, I got the following result

[image]

Have you changed ROS_MASTER_URI and ROS_HOSTNAME to your ip in your bashrc file?
If not, please refer to the [link](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#network-configuration) and edit the bashrc file.
And it would be appreciated if you let us know the test result after running Gazebo

nfhe, 2021-11-23:
Hi, @ashekim

Thank you very much for your reply. 

I have tried your method. After configuring the network, I still encounter the same problem. When I run **roslaunch turnbot3_ gazebo turtlebot3_ AHi, @ashekim

Thank you very much for your reply. 

I have tried your method. After configuring the network, I still encounter the same problem. When I run **roslaunch turnbot3_ gazebo turtlebot3_ After world.launch**, I didn't receive any errors, but there was no robot model in gazebo, and only / clock / rosout / rosout in rostopic list, I hope you can help me solve such a bug again.

**Terminal display**

... logging to /home/he/.ros/log/2c49622c-4c5d-11ec-b511-8c8caa32777f/roslaunch-he-Lenovo-Legion-R7000P2020H-4665.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://192.168.0.111:33505/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [4678]
ROS_MASTER_URI=http://192.168.0.111:11311

setting /run_id to 2c49622c-4c5d-11ec-b511-8c8caa32777f
process[rosout-1]: started with pid [4689]
started core service [/rosout]
process[gazebo-2]: started with pid [4692]
process[gazebo_gui-3]: started with pid [4697]
process[spawn_urdf-4]: started with pid [4701]

Thank you again for your reply.fter world.launch**, I didn't receive any errors, but there was no robot model in gazebo, and only / clock / rosout / rosout in rostopic list, I hope you can help me solve such a bug again.

**Terminal display**

... logging to /home/he/.ros/log/2c49622c-4c5d-11ec-b511-8c8caa32777f/roslaunch-he-Lenovo-Legion-R7000P2020H-4665.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://192.168.0.111:33505/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [4678]
ROS_MASTER_URI=http://192.168.0.111:11311

setting /run_id to 2c49622c-4c5d-11ec-b511-8c8caa32777f
process[rosout-1]: started with pid [4689]
started core service [/rosout]
process[gazebo-2]: started with pid [4692]
process[gazebo_gui-3]: started with pid [4697]
process[spawn_urdf-4]: started with pid [4701]

Thank you again for your reply.

ROBOTIS-Will, 2021-11-24:
Hi @nfhe 

I'm afraid that our documentation is written based on an official Ubuntu + ROS environment, and we can't guarantee whether our test results will work or not under specific environments such as conda.
I'd recommend to search for more resources in the community for running ROS in conda environment.

Thank you.

nfhe, 2021-11-24:
Hi， @ROBOTIS-Will 

Thank you very much for your reply. I will try to use other models on gazebo. If there are new results, I will feed back the results to you. 

thank you.

nfhe, 2021-11-28:
@ashekim 
Hi,again.

I have solved the problem that the robot model cannot be displayed in gazebo, there is no error and the problem of protocol version conflict. I uninstalled protocol 3.15.0 and installed protocol 3.13.0. This problem is solved.

Thank you again for answering so many questions.

ashekim, 2021-11-29:
Hi. @nfhe 
I would greatly appreciate it.
I Hope you have fun with turtlebot3.
And have a nice day! :)

closed 2021-11-29 by ashekim