ROBOTIS-GIT/turtlebot3_simulations #176: TB3 isn't rendered correctly in Gazebo

ljaniec, 2021-11-23:
## Bug report
- operating system: Ubuntu 20.04
- ROS2 Galactic (binaries)

#### Steps to reproduce:
- install `aws-robomaker-small-warehouse-world` and `nav2-simple-commander` ROS2 Galactic packages (binaries)
- `ros2 launch nav2_simple_commander inspection_demo_launch.py`

Part of my `.bashrc` file:
```
source ~/ros2_ws/install/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/ljaniec/ros2_ws/install/aws_robomaker_small_warehouse_world/share/aws_robomaker_small_warehouse_world/worlds:/home/ljaniec/ros2_ws/install/aws_robomaker_small_warehouse_world/share/aws_robomaker_small_warehouse_world/models:/home/ljaniec/ros2_ws/install/neo_simulation2/share/neo_simulation2/robots:/home/ljaniec/ros2_ws/install/neo_simulation2/share/neo_simulation2/models:/opt/ros/galactic/share/turtlebot3_gazebo/models:/home/ljaniec/.gazebo/models
```

#### Expected behavior

The simulation with one visible TB3 rendered and Nav2 stack + demo script start.

#### Actual behavior

The simulation doesn't have correctly rendered robot (screens below). Autonomy demo still works (robot moves, LiDAR can be seen etc).
-

[image]

[image]

[image]

[image]

#### Additional information

What I tried:

- reinstall Gazebo and Nav2 packages

- I have found a similar bug [there](https://github.com/mkrizmancic/franka_gazebo/issues/1) 

- I checked URDF like [there](http://gazebosim.org/tutorials/?tut=ros_urdf) and it didn't show problems

- this bug shows in this package too (`nav2_rosdevday_2021`, [there](https://github.com/SteveMacenski/nav2_rosdevday_2021/issues/7) ), one with MP 400 (the robot is invisible, in "Collisions" view in Gazebo too - unlike there with TB3) - maybe it is something connected with these meshfile changes mentioned in the first link above?

Am I doing something wrong there? Standard simulation (1 TB3 in TurtleWorld) is working fine - robot model loads etc. 
The discrepancy between simulations with the same model (AFAIK) is surprising to me.

ROBOTIS-Will, 2021-11-24:
Hi @ljaniec 

The turtlebot3_simulations package became independent from the turtlebot3 package and had to keep separate modeling files.
The modeling files in the simulations package has been extremely simplified in order to minimize the size of the package which had been requested from several developers.

In order to use more realistic design of the robot, you may use the mesh files in the turtlebot3_description package.
https://github.com/ROBOTIS-GIT/turtlebot3/tree/master/turtlebot3_description/meshes

I'll try to come up with a solution that can easily switch the modeling files without increasing the size of the default package.
Please let me know if you have any thoughts for this.

Thanks!

ljaniec, 2021-11-29:
Hello,

I prepared a small set of commands to show, what I think the problem is.

## TB3 case 1

### Start Gazebo simulation with empty world

`ros2 launch gazebo_ros gazebo.launch.py`

### Add TB3 SDF model to Gazebo (package turtlebot3_gazebo)

`ros2 run gazebo_ros spawn_entity.py -file /opt/ros/galactic/share/turtlebot3_gazebo/models/turtlebot3_waffle/model.sdf -entity robot01`

It spawns TB3 Waffle normally (visible model and LiDAR).

## TB3 case 2-1

### Start Gazebo simulation with empty world

`ros2 launch gazebo_ros gazebo.launch.py`

### Add TB3 URDF model to Gazebo (package nav2_bringup)

`ros2 run gazebo_ros spawn_entity.py -file /opt/ros/galactic/share/nav2_bringup/urdf/turtlebot3_waffle.urdf -entity robot01`

Surprisingly, it spawns invisible TB3 Waffle. It is visible only in the 'Collision' view in Gazebo. No LiDAR too.

## TB3 case 2-2

### Start Gazebo simulation with empty world

`ros2 launch gazebo_ros gazebo.launch.py`

### Generate TB3 SDF model from URDF

`gz sdf -p turtlebot3_waffle.urdf > turtlebot3_waffle.sdf`

### Add TB3 SDF model to Gazebo (from package nav2_bringup)

`ros2 run gazebo_ros spawn_entity.py -file ~/Desktop/turtlebot3_waffle.sdf -entity robot01`

TB3 is invisible (but it's there, after launching 'Collision' view in Gazebo).

## MP400 case

### Start Gazebo simulation with empty world

`ros2 launch gazebo_ros gazebo.launch.py`

### Add MP400 URDF model to Gazebo

`ros2 run gazebo_ros spawn_entity.py -file ~/mp_400/mp_400.urdf -entity robot01`

Robot is invisible, there are only blobs in the air and a LiDAR.

### Generate SDF model from URDF

`gz sdf -p mp_400.urdf > mp_400.sdf`

### Add MP400 SDF model to Gazebo

`ros2 run gazebo_ros spawn_entity.py -file ~/mp_400/mp_400.sdf -entity robot01`

Robot is again invisible, there are only blobs in the air and a LiDAR.

## What is the problem there and how do I fix it?

Relevant issues and links I have found:
* https://github.com/ros-planning/navigation2/issues/2157 - but it's is solved - there is `/opt/ros/galactic/share/turtlebot3_gazebo` already
* https://answers.gazebosim.org//question/26073/cannot-spawn-urdf-into-gazebo-using-ros2/ and https://answers.gazebosim.org//question/2282/convert-urdf-to-sdf-or-load-urdf/ - these two 
* https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1284 - it looks like this could be the problem addressed it this (unmerged yet) PR ("[gazebo_ros_factory]: Resolve package:// urls")

I have all used files there:
* [MP400](https://github.com/neobotix/neo_simulation2/tree/galactic/robots/mp_400) from [Neobotix](https://github.com/neobotix/neo_simulation2), branch `galactic`
* [TB3](https://github.com/ros-planning/navigation2/tree/galactic/nav2_bringup/bringup/urdf) from Nav2 repository, branch `galactic`

ljaniec, 2021-12-11:
Relevant issue: https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1272#issuecomment-857513366

sunghowoo, 2025-02-04:
This issue is too old, and currently, TurtleBot will only support Noetic and Humble. I will close this issue.

closed 2025-02-04 by sunghowoo