ROBOTIS-GIT/turtlebot3_simulations #179: Add Ignition Gazebo support

chapulina, 2022-01-28:
[Ignition Gazebo](https://ignitionrobotics.org) is the latest generation of the Gazebo simulator. This repository only supports Gazebo classic so far, and that's going to EOL in 2025. It would be nice to give users an Ignition simulation and tutorials as well, so they can start migrating before then.

Ignition Gazebo is supported on most currently supported ROS distros. See [REP-003](https://ros.org/reps/rep-0003.html) and [REP-2000](https://ros.org/reps/rep-2000.html).

* Melodic: No official Ignition support, but there's support compiling from source
* Noetic: Uses Igntion Citadel
* Foxy: Uses Ignition Citadel
* Galactic: Uses Ignition Edifice
* Rolling: Uses Ignition Fortress
* Humble: Will use Ignition Fortress

Open Robotics can do the work of adding a `turtlebot3_ignition` package or Noetic, Foxy, Galactic and Rolling. It could live alongside the existing `turtlebot3_gazebo`, so that can continue to be supported until Gazebo classic reaches end-of-life.

I expect `turtlebot_ignition` to be a thin package that uses the [ros_ign_bridge](https://github.com/ignitionrobotics/ros_ign/) to pass ROS commands directly to the simulator and read sensor data directly.

Before we open pull requests, we'd like to check if there's interest from the maintainers in accepting the contribution.

ROBOTIS-Will, 2022-02-03:
Hi @chapulina 

Thanks for the heads up.
Yes, we're willing to accept the changes that add the Ignition Gazebo in the `turtlebot3_simulations` package.

@ashekim 
Please review and test when the PR arrives. We may also need to update our eManual.

ahcorde, 2022-02-04:
Hi @ROBOTIS-Will and @ashekim 

I created a draft PR https://github.com/ROBOTIS-GIT/turtlebot3_simulations/pull/180 with Ignition Gazebo support. I tested it on Galactic. 

Happy to get some feedback

ROBOTIS-Will, 2022-02-07:
@ahcorde 
Thanks for your contribution!
We'll merge first and test on our local machines.

ashekim, 2022-02-10:
@ahcorde 
Thanks for your contribution :)
I'll review and test it on my local machine

Thank you and have a nice day!

TannerGilbert, 2024-12-13:
Sorry to bring up this old issue but I'm quite interested in continuing using the Turtlebot 3 simulation even now that I switched to Gazebo Harmonic. The PR by @ahcorde looks promising but the branch it was merged into doesn't exist anymore. Are there any plans to support newer versions of Gazebo for the turtlebot3_simulation or should one use Turtlebot 4 instead?

For reference currently the best option (in my opinion) for using the Turtlebot3 with Gazebo >Fortess is the fork by @azeey: https://github.com/azeey/turtlebot3_simulations/tree/new_gazebo (also see the Gazebo migration guide: https://gazebosim.org/docs/latest/migrating_gazebo_classic_ros2_packages/)

TZECHIN6, 2025-01-06:
I have tried to self implement the migration following this official tutorial https://gazebosim.org/docs/fortress/migrating_gazebo_classic_ros2_packages/, and also tried out the fork provided https://github.com/azeey/turtlebot3_simulations/tree/new_gazebo...

The simulation can be launch without issue, but I found some errors exist in the simulation:
1. I believe the tf is broken, the lidar frame is at [0,0,0] instead of the expected base_scan frame.
2. Also when using `teleop_twist_keyboard` to publish `cmd_vel` to control the sim robot, even through the robot move, but there is a ghost robot stay at [0,0,0] as well.

I am using ROS2 Humble + Ignition Fortress + Ubuntu 22, could anyone help inspect this issue?

TannerGilbert, 2025-01-06:
@TZECHIN6 the fork that I mentioned above should not have the issues you mentioned above (I tested it with ROS2 Humble and Gazebo Harmonic). The only thing I think is still needed to get that fork into the main repo is that not all launch files were adapted to the new Gazebo and it's also a question if the Gazebo Classic support should be dropped (at least inside main) or if a separate package for newer Gazebo versions is desired.

TZECHIN6, 2025-01-07:
[image]

[image]

I have used another computer to rebuild the example, now the ghost robot is gone when driving it around, thats good, however as you see in the picture, the laser scan rendering still got some issues, it stick with the [0,0,0], even through it will update reflected ray,  but its a off... The tf graph seem fine to  me, so I think its a rendering issue?

Btw, why you use Harmonic instead of Fortress, as the official document suggest using Fortress if Humble is selected.

TZECHIN6, 2025-01-13:
I finally figure it why, if <gz_frame_id>base_scan</gz_frame_id> is set, it will break the tf I believe, and the visualization would always stay at 0,0,0 position instead of the chassis link (or base_link)

azeey, 2025-01-13:
@TZECHIN6 there's a bug affecting this in Fortress. The fix has just been released https://github.com/gazebosim/gz-sim/pull/2712. Can you try updating to ign-gazebo 6.17?

TZECHIN6, 2025-01-14:
@azeey just got an update from apt, 6.17 version fix this bug. :+1:

sunghowoo, 2025-02-04:
For the latest Gazebo version support, our team will be formed in 2025, and we plan to provide support for Gazebo Sim and Jazzy versions within Q2. I will close this issue.

closed 2025-02-04 by sunghowoo