ROBOTIS-GIT/turtlebot3_simulations #183: C++ language error while compiling src/turtlebot3_simulations/turtlebot3_gazebo/src/turtlebot3_drive.cpp

2catycm, 2022-03-13:
## What happens?

```bash
[ 98%] Building CXX object turtlebot3_simulations/turtlebot3_gazebo/CMakeFiles/turtlebot3_drive.dir/src/turtlebot3_drive.cpp.o
In file included from /mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/class.h:28,
                 from /mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/object.h:22,
                 from /mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectimpl.h:21,
                 from /mnt/c/opt/ros/foxy/x64/include/log4cxx/level.h:24,
                 from /opt/ros/noetic/include/ros/console.h:46,
                 from /opt/ros/noetic/include/ros/ros.h:40,
                 from /mnt/d/EnglishStandardPath/Practice_File/P_Intelligent_Robotics/lab4_ws/src/turtlebot3_simulations/turtlebot3_gazebo/include/turtlebot3_gazebo/turtlebot3_drive.h:22,
                 from /mnt/d/EnglishStandardPath/Practice_File/P_Intelligent_Robotics/lab4_ws/src/turtlebot3_simulations/turtlebot3_gazebo/src/turtlebot3_drive.cpp:19:
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:42:30: error: expected initializer before ‘ObjectPtrBase’
   42 |         class LOG4CXX_EXPORT ObjectPtrBase {
      |                              ^~~~~~~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:54:9: error: expected class-name before ‘{’ token
   54 |         {
      |         ^
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:82:25: error: ‘ObjectPtrBase’ does not name a type; did you mean ‘ObjectPtrT’?
   82 |        ObjectPtrT(const ObjectPtrBase& p1)
      |                         ^~~~~~~~~~~~~
      |                         ObjectPtrT
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:82:8: error: ‘log4cxx::helpers::ObjectPtrT<T>::ObjectPtrT(const int&)’ cannot be overloaded with ‘log4cxx::helpers::ObjectPtrT<T>::ObjectPtrT(const int&)’
   82 |        ObjectPtrT(const ObjectPtrBase& p1)
      |        ^~~~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:56:10: note: previous declaration ‘log4cxx::helpers::ObjectPtrT<T>::ObjectPtrT(const int&)’
   56 |          ObjectPtrT(const int& null)
      |          ^~~~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:89:32: error: expected ‘)’ before ‘&’ token
   89 |        ObjectPtrT(ObjectPtrBase& p1)
      |                  ~             ^
      |                                )
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:141:29: error: declaration of ‘operator=’ as non-function
  141 |       ObjectPtrT& operator=(ObjectPtrBase& p1) {
      |                             ^~~~~~~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:141:27: error: expected ‘;’ at end of member declaration
  141 |       ObjectPtrT& operator=(ObjectPtrBase& p1) {
      |                           ^
      |                            ;
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:141:42: error: expected ‘)’ before ‘&’ token
  141 |       ObjectPtrT& operator=(ObjectPtrBase& p1) {
      |                            ~             ^
      |                                          )
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:146:35: error: ‘ObjectPtrBase’ does not name a type; did you mean ‘ObjectPtrT’?
  146 |       ObjectPtrT& operator=(const ObjectPtrBase& p1) {
      |                                   ^~~~~~~~~~~~~
      |                                   ObjectPtrT
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:146:19: error: ‘log4cxx::helpers::ObjectPtrT<T>& log4cxx::helpers::ObjectPtrT<T>::operator=(const int&)’ cannot be overloaded with ‘log4cxx::helpers::ObjectPtrT<T>& log4cxx::helpers::ObjectPtrT<T>::operator=(const int&)’
  146 |       ObjectPtrT& operator=(const ObjectPtrBase& p1) {
      |                   ^~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:116:22: note: previous declaration ‘log4cxx::helpers::ObjectPtrT<T>& log4cxx::helpers::ObjectPtrT<T>::operator=(const int&)’
  116 |          ObjectPtrT& operator=(const int& null) //throw(IllegalArgumentException)
      |                      ^~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h: In constructor ‘log4cxx::helpers::ObjectPtrT<T>::ObjectPtrT(const int&)’:
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:58:17: error: ‘ObjectPtrBase’ has not been declared
   58 |                 ObjectPtrBase::checkNull(null);
      |                 ^~~~~~~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h: In constructor ‘log4cxx::helpers::ObjectPtrT<T>::ObjectPtrT(const int&)’:
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:83:11: error: request for member ‘cast’ in ‘p1’, which is of non-class type ‘const int’
   83 |           _LOG4CXX_OBJECTPTR_INIT(reinterpret_cast<T*>(p1.cast(T::getStaticClass())))
      |           ^~~~~~~~~~~~~~~~~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h: In member function ‘log4cxx::helpers::ObjectPtrT<T>& log4cxx::helpers::ObjectPtrT<T>::operator=(const int&)’:
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:121:17: error: ‘ObjectPtrBase’ has not been declared
  121 |                 ObjectPtrBase::checkNull(null);
      |                 ^~~~~~~~~~~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h: In member function ‘log4cxx::helpers::ObjectPtrT<T>& log4cxx::helpers::ObjectPtrT<T>::operator=(const int&)’:
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h:147:46: error: request for member ‘cast’ in ‘p1’, which is of non-class type ‘const int’
  147 |          T* newPtr = reinterpret_cast<T*>(p1.cast(T::getStaticClass()));
      |                                              ^~~~
/mnt/c/opt/ros/foxy/x64/include/log4cxx/helpers/objectptr.h: In member function ‘T* log4cxx::helpers::ObjectPtrT<T>::exchange(const T*)’:
[... 66 lines cut ...]
 ```bash
git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git 
git clone https://github.com/ROBOTIS-GIT/turtlebot3.git 
git clone https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git 
cd .. && catkin_make
  ```
## ROS env

```bash
ROS_VERSION=1
ROS_PYTHON_VERSION=3
ROS_PACKAGE_PATH=/mnt/d/EnglishStandardPath/Practice_File/P_Intelligent_Robotics/lab4_ws/src:/opt/ros/noetic/share
ROSLISP_PACKAGE_DIRECTORIES=/mnt/d/EnglishStandardPath/Practice_File/P_Intelligent_Robotics/lab4_ws/devel/share/common-lisp
ROS_ETC_DIR=/opt/ros/noetic/etc/ros
ROS_MASTER_URI=http://localhost:11311
ROS_ROOT=/opt/ros/noetic/share/ros
ROS_DISTRO=noetic
```

## OS env
- Windows 11

- wsl2  core 

Linux DESKTOP-SPT1P3R 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

2catycm, 2022-03-13:
As you can see in the error message, a ros2 foxy is installed on windows but not wsl in C:/opt/ros/foxy by using chocolatey. Bash in wsl does not source the devel/setup.bat, but it seems that CMake in wsl is trying to use the log4cxx contained in foxy.

sunghowoo, 2025-02-04:
We currently support ROS 2 Humble, but we do not officially support WSL environments. As a result, debugging this issue would be difficult. We appreciate your understanding. Since this issue has been open for a long time, we will proceed with closing it.

closed 2025-02-04 by sunghowoo