ROBOTIS-GIT/turtlebot3_simulations #184: --- stderr: turtlebot3_gazebo CMake Error at CMakeLists.txt:24 (find_package):   By not providing "Findgazebo_ros_pkgs.cmake" in CMAKE_MODULE_PATH

sam1226, 2022-04-08:
My environment info : win11+wsl2_ubuntu20.04+ROS2-foxy
I want to build the turtlebot3,  download the source code according to turtlebot3.repos and use the version foxy-devel.
but when do the build cmd :colcon build --symlink-install. 
an error occurs : 
*******************************************************************************************************
--- stderr: turtlebot3_gazebo                             
CMake Error at CMakeLists.txt:24 (find_package):
  By not providing "Findgazebo_ros_pkgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gazebo_ros_pkgs", but CMake did not find one.

  Could not find a package configuration file provided by "gazebo_ros_pkgs"
  with any of the following names:

    gazebo_ros_pkgsConfig.cmake
    gazebo_ros_pkgs-config.cmake

  Add the installation prefix of "gazebo_ros_pkgs" to CMAKE_PREFIX_PATH or
  set "gazebo_ros_pkgs_DIR" to a directory containing one of the above files.
  If "gazebo_ros_pkgs" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< turtlebot3_gazebo [18.6s, exited with code 1]
********************************************************************************************************
I check the source code downloaded, and can't find the gazebo_ros_pkgs, and also in repos foxy-devel. should  i download gazebo_ros_pkgs from other repos version to fix this error?
plz help me!

ROBOTIS-Will, 2022-04-12:
Hi,
It looks like you have not installed the gazebo_ros_pkgs package.
https://answers.ros.org/question/349782/why-is-not-building-package-turtlebot3_gazebo-on-ros2/

sunghowoo, 2025-02-04:
This issue has been inactive for too long, so we will proceed with closing it.

closed 2025-02-04 by sunghowoo