ROBOTIS-GIT/turtlebot3_simulations #187: Multi TurtleBot3 Simulation in Gazebo

MPTG94, 2022-04-30:
Hello,

I am trying to run a TB3 simulation where I have 2 or more Burger robots running at the same time.

I tried modifying the existing `turtlebot3_world.launch.py` package in the following manner:

```
import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration

def generate_launch_description():
    launch_file_dir = os.path.join(get_package_share_directory('turtlebot3_gazebo'), 'launch')
    pkg_gazebo_ros = get_package_share_directory('gazebo_ros')

    use_sim_time = LaunchConfiguration('use_sim_time', default='true')
    x_pose = LaunchConfiguration('x_pose', default='-2.0')
    y_pose = LaunchConfiguration('y_pose', default='-0.5')
    # I added these 2 new initial poses
    x_2_pose = LaunchConfiguration('x_pose', default='-5.5')
    y_2_pose = LaunchConfiguration('y_pose', default='-0.5')

    world = os.path.join(
        get_package_share_directory('turtlebot3_gazebo'),
        'worlds',
        'turtlebot3_world.world'
    )

    gzserver_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(pkg_gazebo_ros, 'launch', 'gzserver.launch.py')
        ),
        launch_arguments={'world': world}.items()
    )

    gzclient_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(pkg_gazebo_ros, 'launch', 'gzclient.launch.py')
        )
    )

    robot_state_publisher_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(launch_file_dir, 'robot_state_publisher.launch.py')
        ),
        launch_arguments={'use_sim_time': use_sim_time}.items()
    )

    spawn_turtlebot_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(launch_file_dir, 'spawn_turtlebot3.launch.py')
        ),
        launch_arguments={
            'x_pose': x_pose,
            'y_pose': y_pose
        }.items()
    )

    # I added this new robot spawner which uses the new poses I added
    spawn_turtlebot2_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(launch_file_dir, 'spawn_turtlebot3.launch.py')
        ),
        launch_arguments={
            'x_pose': x_2_pose,
            'y_pose': y_2_pose
        }.items()
    )

    ld = LaunchDescription()

    # Add the commands to the launch description
    ld.add_action(gzserver_cmd)
    ld.add_action(gzclient_cmd)
    ld.add_action(robot_state_publisher_cmd)
    ld.add_action(spawn_turtlebot_cmd)
   # I added an action to use my additional spawner
    ld.add_action(spawn_turtlebot2_cmd)

    return ld

```

I then used colcon build to build the updated pacakge (saved under a file called `turtlebot3_2_world.launch.py`)
And tried running it using the command:
`ros2 launch turtlebot3_gazebo turtlebot3_2_world.launch.py`

The Gazebo software does start with the world map, but it only has one Turtlebot3 model in it, looking at the output of the terminal I am launching turtlebot3 gazebo from I can see some new error messages:

```
[spawn_entity.py-5] [INFO] [1651328573.694735640] [spawn_entity]: Spawn Entity started
[spawn_entity.py-5] [INFO] [1651328573.694987846] [spawn_entity]: Loading entity XML from file /home/mor/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-5] [INFO] [1651328573.695540106] [spawn_entity]: Waiting for service /spawn_entity, timeout = 5
[spawn_entity.py-5] [INFO] [1651328573.695748298] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1651328573.697360962] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1651328573.697610798] [spawn_entity]: Loading entity XML from file /home/mor/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-5] [INFO] [1651328573.697874998] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1651328573.698375179] [spawn_entity]: Waiting for service /spawn_entity, timeout = 5
[spawn_entity.py-4] [INFO] [1651328573.698586519] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1651328573.700757510] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-5] [INFO] [1651328573.717435637] [spawn_entity]: Spawn status: Entity [burger] already exists.
[spawn_entity.py-5] [ERROR] [1651328573.717667760] [spawn_entity]: Spawn service failed. Exiting.
[spawn_entity.py-4] [INFO] [1651328573.737101934] [spawn_entity]: Spawn status: Entity [burger] already exists.
[spawn_entity.py-4] [ERROR] [1651328573.737331030] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-5]: process has died [pid 15291, exit code 1, cmd '/opt/ros/galactic/lib/gazebo_ros/spawn_entity.py -entity burger -file /home/mor/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x -2.0 -y -0.5 -z 0.01 --ros-args'].
[ERROR] [spawn_entity.py-4]: process has died [pid 15289, exit code 1, cmd '/opt/ros/galactic/lib/gazebo_ros/spawn_entity.py -entity burger -file /home/mor/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x -2.0 -y -0.5 -z 0.01 --ros-args'].

```

Is there something I am doing wrong? Is there a better way to get more than one turtlebot3 robot in the same simulation?

faisalam1n, 2022-08-30:
I'm not sure about the approach you are trying but I have spawned multiple robots by tweaking the launch file.

sunghowoo, 2025-02-04:
We are preparing to release support for multi-robot usage with ROS 2 Humble within this week. Since this issue has been inactive for too long, we will proceed with closing it.

closed 2025-02-04 by sunghowoo