ROBOTIS-GIT/turtlebot3_simulations #188: Gazebo simulation of Turtlebot3 - problem with showing /scan in rviz

pitosalas, 2022-05-20:
I do this: 

`$ roslaunch turtlebot3_gazebo turtlebot3_stage_1.launch`

and I see this:

```
started roslaunch server http://10.42.1.247:34485/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /use_sim_time: 0.1

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://10.42.1.247:11311

process[gazebo-1]: started with pid [9153]
process[gazebo_gui-2]: started with pid [9158]
process[spawn_urdf-3]: started with pid [9163]
[ INFO] [1653081915.367759205]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1653081915.368915738]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1653081915.440516461]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1653081915.441538179]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1653081915.635600271]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1653081915.647665343]: Physics dynamic reconfigure ready.
[ INFO] [1653081916.372409506]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1653081916.375630927]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1653081916.482974171]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1653081916.483037100]: Starting Laser Plugin (ns = /)
[ INFO] [1653081916.483665165]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1653081916.583424184]: Starting plugin DiffDrive(ns = //)
[ INFO] [1653081916.583479709]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1653081916.584163860]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1653081916.584763813]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1653081916.585176271]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1653081916.586702853]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1653081916.587067601]: DiffDrive(ns = //): Advertise odom on odom 
[spawn_urdf-3] process has finished cleanly
log file: /my_ros_data/.ros/log/3239ce14-d881-11ec-89f4-f60f84c781c6/spawn_urdf-3*.log
```
Everything looks ok except the last line `[spawn_urdf-3] process has finished cleanly` is a bit suspicious.

Then I do:

`$ roslaunch turtlebot3_gazebo turtlebot3_gazebo_rviz.launch
`
And that launches rviz. However here's where it gets weird. If I display /scan and set the fixed Frame to base_link, then /scan looks correct. But if I change Fixed Frame to odom then the /scan has an error. 

I suspect something with the simulated odom or the simulated tfs but I thought I would ask here.

ROBOTIS-Will, 2022-05-30:
Hi @pitosalas 

The turtlebot3_stage_1.launch is designed for running on [Machine Learning example](https://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#machine-learning).
You may need to install all the necessary environment and set up parameters to properly run the stage.
However, unfortunately, the machine learning section is not supported in Noetic.

Thank you.

sunghowoo, 2025-02-04:
This issue has been inactive for too long, so I will proceed with closing it.

closed 2025-02-04 by sunghowoo