ROBOTIS-GIT/turtlebot3_simulations #194: Unable to find uri

jianqin-mok, 2022-09-03:
I was trying out the TurtleBot3 Gazebo simulation in a VirtualBox, Ubuntu 16.04.
Installations following emanual for PC setup + simulation on https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/ 

When I launch Gazebo my TurtleBot burger seems to fall to underground, then I checked my terminal and I think the root cause might be from these two errors.

> Error [parser.cc:581] Unable to find uri[model://sun]
Error [parser.cc:581] Unable to find uri[model://ground_plane]

I am not bothered by the sun but I think I atleast need the ground plane since the gravity in physics properties is not 0.

jianqin-mok, 2022-09-03:
seems like the `~/.gazebo/models` folder that suppose to have all the default gazebo model is empty. I solved mine by manual adding the gound_plane folder into `~/.gazebo/models`.

brean, 2022-11-23:
The Ubuntu 16.04 version is quite outdated, I suggest you switch to a newer version (we teached robotics with ROS melodic on Ubuntu 18.04 and noetic/Ubuntu 20.04 using the turtlebot3-simulation in gazebo in a virtual machine and it loaded the empty-world with the sun and the ground plane correctly)

sunghowoo, 2025-02-04:
Our team currently supports Noetic and ROS 2 Humble, but we do not support EOL versions. Please try with the latest version. Since this issue has been inactive for too long, we will proceed with closing it.

closed 2025-02-04 by sunghowoo